ra3xdh/qucs_s #1012: Simulation console screen position

tomhajjar, 2024-10-21:
I realize this is a trivial Issue but can the Simulation console screen position be moved so I can see the plots without having to close or move the pop-up? Too many times I miss a subtle change in a plot/data because the plot is obscured.

[image]

ra3xdh, 2024-10-21:
It's better to let the system window manager to control the dialog windows position. The manual setting of window position has many disadvantages. Also wait unit #235 will be implemented. This will remove the *Simulation console* window and redirect all messages to the dock.

ra3xdh, 2024-10-22:
@ainst , Yes, you may open a PR

PR #1028 (2024-10-27): Save the position / size of ExternSimDialog on exit.
This is a simple change to store the last used position and size of the "Simulate with external simulator" dialog (class ExternSimDialog) when the user presses the "Exit" button.

This change should close #1012 and work on all platforms but I have only tested it on Linux / Qt6.
closed 2024-10-27 by ra3xdh