ra3xdh/qucs_s #1017: Trans Simulation says it ran to full 10 sec but doesn't

tomhajjar, 2024-10-24:
Testing out an old EDN Design Idea. Seeing if thermal analysis can be added to any circuit using standard BJT Spice device models.

I'm running two Trans simulations:

If TR1=300usec and TR2=3sec, TR1/TR2 Simulations look OK.
If TR1=300usec and TR2=10sec, TR1 Simulation looks OK. TR2 simulation is wrong.

[image]
[image]

[Amp_BJT_thermal_prj.zip](https://github.com/user-attachments/files/17509501/Amp_BJT_thermal_prj.zip)

ivandi69, 2024-10-24:
It works fine. You have to decrease the step (or increase the number of points). Adding .OPTION INTERP reduces the dataset size. Here is the result with tstep=2ms:
[image]

tomhajjar, 2024-10-24:
Something wrong with my syntax...
Can you send file?

[image]

tomhajjar, 2024-10-24:
I'm getting strange plots for a similar circuit. The Tj2 plot for the PNP changes wildly with #points. Attached are for 5,000 and 6,000 points. 

[image]
[image]

[BJT_thermal_prj.zip](https://github.com/user-attachments/files/17511157/BJT_thermal_prj.zip)

ivandi69, 2024-10-24:
Step 50us. Using KLU as Direct Linear Solver:
[image]
[image]

tomhajjar, 2024-10-24:
I don't see KLU...
[image]

closed 2024-10-24 by tomhajjar