ra3xdh/qucs_s #1023: Misleading axis description

dwarning, 2024-10-25:
[image]

The y axis descriptions are misleading. If I make a S-Parameter simulation with noise calculation ON everything in the plotted diagrams get v(), e.g. v(s_2_1). Same for noise figure values and other results. The wisdom is, NF and NFmin comes from ngspice as dB and noise equivalent resistor has Ohm. 
I recommend removing the v() for any s-param result.

ra3xdh, 2024-10-25:
The Qucs-S takes the variable names from RAW dataset as is and has no smart variable recognition. If Ngspice puts the NF parameter as V(NF) it will be plotted as voltage. As far as I know all outputs of S-parameter simulation are represented as virtual voltages.

closed 2024-10-26 by ra3xdh

dwarning, 2024-10-26:
@ra3xdh You are right it is not a bug, perhaps we have to solve it in ngspice.

tomhajjar, 2024-11-06:
@dwarning @ra3xdh 

Will this change to ngspice effect Qucs-S?

https://sourceforge.net/p/ngspice/discussion/127605/thread/8afc68a7c4/

dwarning, 2024-11-06:
No - as you see in the exported picture. The plot vectors are just renamed, got right type and qucs-s can plot them with their genuine name on axis.