ra3xdh/qucs_s #1026: Multiple anomalies..

tomhajjar, 2024-10-26:
I'm trying to help @bnillsson11 with #1025. He sent multiple version of amplifiers. I'm starting with the "easy' one.

I ran into some anomalies that I need to understand.

1) Qucs-S doesn't show "progress" in the Simulation Console for MFA1-2NXX55.sch, no "Reference value:xxxx. The simulation takes so long it appears the simulation "hung" but it isn't. I made MFA1-2NXX55.cir and used DuSpice. DuSpice does show "progress" AND it seems the simulation runs faster...

2) The 2nd Transient simulation, 5sec produces Vout with visible "cross-over" distortion. How is this possible if Temp is fixed and none of the Spice models are "thermal" models.

3) @bnillsson11 is complaining that changing Tambient from 25 to say 27 causes major changes in the amplifier bias and output distortion. I haven't investigated that yet. The amp he sent has other issues to be fixed first.

[image]
[image]
[image]

[Amp_MFA1_prj.zip](https://github.com/user-attachments/files/17528750/Amp_MFA1_prj.zip)

ra3xdh, 2024-10-26:
> Qucs-S doesn't show "progress" in the Simulation Console for MFA1-2NXX55.sch, no "Reference value:xxxx.

It is not a bug. I have tried to run the simulation from the CLI version of Ngspice. It doesn't report the progress too. I cannot show the progress in the GUI if the CLI backend doesn't report it. I have no idea how the DuSpice extracts the progress if Ngspice doesn't print it in the console. Qucs-S doesn't apply any explicit filter to the logs and shows everything as is. 

> The 2nd Transient simulation, 5sec produces Vout with visible "cross-over" distortion.

It is more likely Ngspice issue. The models are transferred to the simulator as is. The CLI version shown the same result. Try to provide a minimal example reproducing the problem.

>  is complaining that changing Tambient from 25 to say 27 causes major changes in the amplifier bias and output distortion. 

Yes, I have tried this simulation. This definetely not a GUI bug. The simulator shows exactly the same that CLI backend outputs. Please try to provide a minimal example reproducing a problem.

ra3xdh, 2024-10-26:
I have noticed that your are trying to set `Integration=Trapezoidal` property on the TRAN simulation. This property has no effect for SPICE mode. The implementation of #974 will prevent such accidental incompatible property setting.

closed 2024-10-26 by ra3xdh

bnilsson11, 2024-10-26:
A comment about the "crossover distorsion":
It is not an anomaly, it is the normal/true behaviour of this circuit, it has been verified in hardware.
Don't spend time on it. It is a circuit design issue, not a simulation issue.

I am sending a cleaned-up version of the circuit that should have no "formal" issues.
The main problem for me right now is that TEMP=27 works, while TEMP=27.1 does not.
[MFA1v8debug_cleaned.zip](https://github.com/user-attachments/files/17529609/MFA1v8debug_cleaned.zip)
To create a minimal example will be a problem or might take time for me since it will not reproduce in a new design, at least not from the beginning. It seems like a medium term "file corruption" problem. But I will try.

My platform, might be important:
MacOS M2 15.0.1 (24A348)
Ngspice 42
Qucs-s 24.3.2

tomhajjar, 2024-10-26:
This was Holger's comment on the Progress metric...

https://sourceforge.net/p/ngspice/discussion/120973/thread/ce5193d280/