ra3xdh/qucs_s #1033: Tabular diagram (table) issues - properties only accessible through double-click

njascgil, 2024-10-30:
Using the latest continuous build of Qucs-S, I noticed that the tabular diagram properties are only accessible through double-click and not through the right-click menu (in the case of the cartesian diagram, both methods work as expected).

Additionally I would like to ask if there's any way to modify value order in these tabular diagrams, other than removing values and re-adding them by the desired order. Thanks!

ra3xdh, 2024-10-30:
> I noticed that the tabular diagram properties are only accessible through double-click 

Yes, I confirm the problem. The RMB than click on "Edit properties" do nothing for Tabular diagram.

> Additionally I would like to ask if there's any way to modify value order in these tabular diagrams

There is no way to reorder data columns in tabular diagram. You have to delete the data and add it again.

ra3xdh, 2024-10-31:
Fixed by #1035

PR #1035 (2024-10-31): Fixed context menu for tab diagram and ASCO execution error
This PR fixes the following bugs:

* Fixes #1033. The problem was the context menu execution on TabDiagram was not allowed. 
* Fixes ASCO execution error on Linux reported by mailing list https://sourceforge.net/p/qucs/mailman/message/58834975/
closed 2024-10-31 by ra3xdh