ra3xdh/qucs_s #1034: Digital Libraries for ngspice

tomhajjar, 2024-10-30:
Issue #900 was getting too long...

@ra3xdh since you are assigning this task to a student I though this info might help. 

ngspice supports multiple Digital Libraries but the ones one from "MicroCap" offer the most devices. These are simple models that support gate delays and I/O voltage levels. More complete models can be found but there are so few the Library would be tiny.

MicroCap_dig000.lib has 74, AC, LS, HC...etc. Unfortunately there are many missing devices.
MicroCap_74lv.lib has a wide selection of 74LV devices.
MicroCap_cmos.lib has CD4000 devices.
MicroCap_ecl.lib has ECL devices.

There may be other MicroCap Digital Libraries. They would need to be tested to work in ngspice.

tomhajjar, 2024-10-31:
I found additional MicroCap Digital Libraries. The following are included above.

dig000.lib
74lv.lib
cmos.lib
ecl.lib

[MicroCap_digital_libs.zip](https://github.com/user-attachments/files/17580844/MicroCap_digital_libs.zip)

tomhajjar, 2024-11-04:
Found a NXP 74HC component-level library. Might be useful for mixed-signal analysis, I used the 74HCU04.

To make individual Spice models requires a lot of copy-paste. A simple device might have 6 sub-circuits throughout the files.

[NXP 74HC_component-level.zip](https://github.com/user-attachments/files/17613069/NXP.74HC_component-level.zip)

ra3xdh, 2024-11-04:
> Found a NXP 74HC component-level library.

Thanks! This would be the best library for analog mode simulations.

tomhajjar, 2024-11-04:
LTspice uses sub-circuits called dviewX making calls to dview.lib to create Timing Diagrams. This is another option.

[image]

[LTspice Digital-View.zip](https://github.com/user-attachments/files/17621276/LTspice.Digital-View.zip)

tomhajjar, 2024-11-26:
@ra3xdh 

Any digital libraries from your students you want me to review?

ra3xdh, 2024-11-26:
There is no progress on this subject yet.

tomhajjar, 2025-01-05:
Latest version of the Digital Libraries.

Digital_CD4000 -> Digital_CD

[Digital Libraries.zip](https://github.com/user-attachments/files/18308944/Digital.Libraries.zip)

ra3xdh, 2025-01-05:
I have uploaded the updated libraries by #1184. This issues may be closed. The additional devices may be uploaded later. The CD4066 device still be missing. It is often used in analog circuits. I will try to finish my transistor level model for this IC and upload the library.

closed 2025-01-05 by ra3xdh

tomhajjar, 2025-01-05:
CD4066 from LTspice. One model tracks "real" life Ron versus supply.

[CD4066.zip](https://github.com/user-attachments/files/18310255/CD4066.zip)

tomhajjar, 2025-01-05:
I modified the above CD4066 model slightly. Did a quick test and added it to the end of Digital_CD.lib. Updated files above.

[image]

[CD4066_prj.zip](https://github.com/user-attachments/files/18310701/CD4066_prj.zip)

ra3xdh, 2025-01-05:
CD4066 model uploaded by #1187. Thanks!

tomhajjar, 2025-01-05:
I decided to add the CD4069UB. Since it's "analog" like the CD4066 it was added to the end of the Library.

Digital_CD.lib updated above.

[image]

ra3xdh, 2025-01-06:
CD4069 model uploaded by #1190