ra3xdh/qucs_s #1037: Quartz Crystal Library

tomhajjar, 2024-10-31:
The new crystal library can be expanded using the attached libraries. I will investigate what devices to add to the library.

[XTAL.zip](https://github.com/user-attachments/files/17588954/XTAL.zip)

tomhajjar, 2024-10-31:
First cut at XTAL library. 

1) I redid "XTAL" to be the same as the 10MHz crystal I chose. I did this because 10MHz is a common crystal unlike 8863KHz and I changed the variable names to be consistent with the Library I chose. XTAL can be treated like a Generic Crystal and allow the User to alter parameters. 

2) ngspice/Qucs does not like any model or subcircuit "names" with a decimal "." in them. Not easy coming up with crystal names without a decimal. Also ngspice won't allow a Spice file to contain ".SUBCIRCUIT XTAL...". The "X" causes many errors.

ra3xdh, 2024-10-31:
The existing Crystal.lib is designed specially for quartz ladder fiters design. These filter are usually used for the hamradio HF bands analog transceivers. It contains only one generic model intentionally. The parameters of the crystal should be measured for example using the NanoVNA device and ACMP software https://github.com/Battosai42/amcp and then substiuted in the simulation. Example of such desing is here: http://w7zoi.net/filters/9megxfils.pdf Providing a set of model for different frequencies will only mislead the user. 

I guess the proposed library is taken from MicroCAP distribution or something similar. The model parameters of the XTALs from this library don't match my measurements. Usually it gives Ls around 20mH for HC-49/U crystal at 5-10 MHz. The Ls may differ in range about 15-25mH for crystals with the same package and frequency but different origin. The calculation from library gives Ls in range from 0.8 to 2mH. The Rs seems to be too high. Probably the library was designed for oscillators. 

So, I would keep the existing Crystal.lib as is with only one generic model.

tomhajjar, 2024-10-31:
Agreed. I found a few quarz.sch on the web that had errors. I had some in old projects.

Since the majority of people will never design an XTAL filter, a generic XTAL library would be a good addition.

tomhajjar, 2024-11-05:
I created a new Library called "Crystals.lib". 

It has one "Generic" 10MHz crystal called "XTAL" so Users can set their own parameters.

It also has 12 crystals with fixed parameters. Finding and choosing parameters was not easy. Manufacturers don't publish many and the parameters can vary wildly for the same frequency and type crystal.

I avoided issues with decimals in names by using "_"

[image]

[image]

[XTAL_prj.zip](https://github.com/user-attachments/files/17633126/XTAL_prj.zip)

ra3xdh, 2024-12-29:
I have uploaded this library here: https://github.com/ra3xdh/qucs-rus-complib I am not inclined to include the library that doesn't reflect the measured parameters. I will try to measure more quartz crystals in different packages (HC-49U, HC-49S etc.) and update the library in distribution. The crystals should be sorted by package. I will perform this task later. Maybe assign to one of my students.

closed 2024-12-29 by ra3xdh