ra3xdh/qucs_s #1040: I PWL icon arrow inverted?

tomhajjar, 2024-11-01:
Is the I PWL icon arrow inverted? It's the only "Red" Current Source in the menu with the arrow pointing up.

I haven't tested the "Blue" sources in a while but I seem to remember a polarity issue in the past...

[image]

[I-PWL.zip](https://github.com/user-attachments/files/17601902/I-PWL.zip)

tomhajjar, 2024-11-02:
No constancy for the arrows.

[image]

[I _Source_test.zip](https://github.com/user-attachments/files/17607914/I._Source_test.zip)

ra3xdh, 2024-11-12:
The sources polarity fixed by #1059

closed 2024-11-12 by ra3xdh