ra3xdh/qucs_s #1045: Add possibility to define the simulation from the main menu

ra3xdh, 2024-11-05:
More than 20 years of Qucs usage showed that the most of users don't understand the "everything is component" concept. And nobody reads the documentation. There exists a demand on the massive GUI redesign #352 I am not considering this option in the near future because there is no full time developers in the project. And the Qucs-S is not supported by any organization like KiCAD. But something could be done by sufficient little effort. 

I propose the following addition to the existing simulation system:

* Add submenu containing menu pointls like *Transisent*, *AC*, *S-parameter*, *DC sweep*, etc. in the *Simulation* menu
* If schematic contains no simulation, the user clicks on this menu and the simulation properties dialog window appears. Once the user clicks OK, the simulation component is automatically placed on the free space in the schematic.
* If shcematic already contains a simulation and user clicks the menu, the application finds the simulation component and open the properties dialog.
* Maybe add the *Run simulation* button in the simulation properties dialog. 

This is a long term development task. I am not planning this for the v25.1.0

iwbnwif, 2024-11-05:
I am working on some refactoring of the Component Properties dialog. I haven't added it to the discussion yet because I am not sure that I will actually achieve what I am setting out to do, which is basically make the editing of parameters similar to QucsStudio.

I don't think that this clashes with your proposal here, but thought I would let you know just in case.

As an aside, I have experimented (with some success) at making the equation function of the of the Component Properties dialog into a code editor. My idea is to have syntax highlighting and eventually code completion. Again, it is very tentative at this stage so only a 'mention' for now...

iwbnwif, 2024-11-05:
A thought on your proposal above. How about the simulation button launching a simulation wizard if no simulation exists? I haven't played with Qt wizards but from my time on WxWidgets I think they are quite straightforward.

zergud, 2024-11-05:
I think that "everything is component" is wellknown concept... from Serenade (RIP) to ADS
there is no need to teach bad practices of “never read documentation”
possible "fast" solution - disable simulation button when no simulation found on schematic :)

instead we can pay attention to #235 
+ may be, rethink simulation workflow - dont select simulator from combobox,  simulator is property of simulation block 
e.g. "Transient (ngspice)" "Transient (xyce)" 
and implement parallel execution (even for different simulation types for one simulator)

zergud, 2024-11-05:
diagram on schematic also "inspired by ADS" and I think its very usefull for RF community 

Anyway, simulation wizard may be usefull but not very easy to implement - many simulation types, different simulators....

iwbnwif, 2024-11-05:
> I am planning the refactoring of the Property class which may involve some changes in componentdialog.cpp. Please keep an eye on the status of https://github.com/ra3xdh/qucs_s/issues/974 when implementing the new features in componentdialog.

Will do. Also, if you change the Component class at all, it would be helpful to include a flag to say if it has a sweep. There is already `isSimulation` but as far as I can tell the only way to detect if it sweeps is to check the Model type.

ra3xdh, 2024-11-06:
> if you change the Component class at all

I am not planning to change anything in Component class. The purpose is to automatically show/hide incompatible properties in the component dialog. I will try to speed up with this feature, so you can start the refactoring.

csrabak, 2024-11-06:
I'll skip the comment about people "not reading the documentation", and just point to some simulators have a very similar HMI to the proposed submenu in this issue.

I concur @iwbnwif's idea about having the simulation button (or menu entry click) checking if no simulation selected.

ra3xdh, 2024-11-22:
Moving to Discussion/Ideas as a low priority task. May be reopened if some developer for this task will be found.