ra3xdh/qucs_s #1046: Waveform/Source Library

tomhajjar, 2024-11-06:
Would it be useful to have a Library of Sources made up of sub-circuits? It would speed making say Triangle, Sawtooth...etc.

[image]
[image]
[image]

[Waveforms_prj.zip](https://github.com/user-attachments/files/17641466/Waveforms_prj.zip)

ra3xdh, 2024-11-06:
Please wait with this library and similar libraries. We need first to backport this rejected PR from Qucs: https://github.com/Qucs/qucs/issues/659 This will allow to show XML defined devices on the system components panel. Otherwise these source will have bad visibility for user.

ra3xdh, 2024-11-08:
I propose the following addition to improve the pulsed source usability:

* Add a button named like *Interactive source setup* in a pulse sources properties dialog. This button will be inactive for other devices. The dialog window will appear after clicking on this button. This dialog will allow to define the parameters based on some patterns like sawtooth, triangle, meander, trapezoidal. The dialog window may contain an image representing waveform, drop-down list and input fields. This will automatically calculate the pulsed source parameters. 
* Alternative: add a special pulse source like *Unified source*. A dialog window like mentioned before will be opened after clinking on this source. This source could be mapped to SPICE `PULSE` source.

In other words this issue should be resolved in C++ level.

tomhajjar, 2024-11-08:
Pulse is a good one to start with. Pulse can be V or I.

One can also specify how many pulses or continuous.

PULSE(V1 V2 TD TR TF PW PER NP)

Name Parameter Default Value Units
V1 Initial value - V, A
V2 Pulsed value - V, A
TD Delay time 0.0 sec
TR Rise time TSTEP sec
TF Fall time TSTEP sec
PW Pulse width TSTOP sec
PER Period TSTOP sec
NP Number of Pulses (NP set to 0 or omitted denotes unlimited pulses)

[image]
[image]

tomhajjar, 2025-01-06:
I made a number of Source subcircuits for use in "analog" and XSPICE "digital" Simulations. Might be useful to someone.

[Waveforms_prj.zip](https://github.com/user-attachments/files/18320036/Waveforms_prj.zip)

ra3xdh, 2025-01-06:
> I made a number of Source subcircuits for use in "analog" and XSPICE "digital" Simulations. Might be useful to someone.

I am not inclined to include this library in the distribution. This problem should be resolved in C++ level. See my comment about a custom dialog above.

tomhajjar, 2025-01-06:
No need to be in distribution or made into a Library. They are available in the Discussion section.

XSPICE will require something different than your planning. XSPICE has built-in digital sources and any analog source can be made into a digital source with a ADC Bridge. I included a number of "digital" sources.