ra3xdh/qucs_s #1047: Show XML defined devices on the left panel.

ra3xdh, 2024-11-06:
This is a long term development task for the next year. There exists a rejected PR for old Qucs: https://github.com/Qucs/qucs/issues/659 It allows to show XML defined devices on the left panel.  This will reduce the compile time and allow to get rid of the most C++ hardcoded devices. The simulations and some other components must remain on the C++ level. Some SPICE devices requires to make computations and look into the parent schematic to generate the netlist entry. But the most of C++ devices could be migrated to XML definition. 

The following tasks must be resolved:

* It's need to define what XML to use. Qucs XML or true XML. Maybe accept Caned library XML format: https://github.com/Caneda/Caneda
* It's need to align this with the planned task for adding PDKs support: #876

KrzysztofHerman, 2024-11-06:
@ra3xdh great to hear this news. It would be a great to make it done. Standardizations also sounds good.
I do not know if it is possible but would be great to define a symbol with standard geometry (you have right now standard geometries hardcoded in C++ in the `microelectronics` section) and instantiation scheme (with some default parameters). 
Also a feature to have kind of checks if a parameter is within a certain range would be more than welcome.

ra3xdh, 2024-11-10:
@felix-salfelder I have read your proposal about unified schematic file format here: https://github.com/Qucs/qucs/issues/1094 At the current time nothing is implemented except the intention to develop a new format. A lot of work is ahead. And you have to implement everything from scratch. I have no wish to participate in the new schematic format implementation (C++ level) development. I would wish you good luck with this task.  I don't consider the changing of existing schematic format in Qucs-S at the moment. Maybe only switching the markup language to the true XML in the long perspective. Surely not to the format that doesn't exists yet.

felix-salfelder, 2024-11-10:
On Sun, Nov 10, 2024 at 08:21:50AM -0800, Vadim Kuznetsov wrote:
> At the current time nothing is implemented except the intention to
> develop a new format.

Thanks Vadim, I can see that. A new format is not needed. We will use a
well designed and proven one.

andresmmera, 2025-04-05:
@ra3xdh , please assign @ThomasZecha to this task instead of me. AFAIK, he's working on it, and his development is more advanced than mine.

I have stopped my work to avoid duplicating efforts.

ThomasZecha, 2025-04-15:
@ra3xdh I have a first running prototyp for XML-based devices. Can you setup a dev-branch for further development?

ra3xdh, 2025-04-16:
@ThomasZecha Why not to prepare #1308 for merging into the `current` branch?

At this point we also have to determine which devices will be migrated to XML and which devices must remain in C++. It is not a straightforward task. For example I am working on magnetic cores devices which requires a custom netlist entry processing and custom properties dialog. Also the custom dialog is planned for pulsed sources #1046 The primitive semiconductor devices have the *Fill from SPICE model* feature. Will this work with XML device? My proposal is the following.

What should be migrated to XML:

* The *Microelectronics* group
* *Digital* group
* SPICE-only sources (blue and red ones)

What should remain in C++:

* Simulations and equations
* File devices
* SPICE netlist sections
* Probes
* Lumped components group (?)

ra3xdh, 2025-04-29:
I have created `dev/xml_devices` branch. @ThomasZecha , you may retarget the #1307 to this branch.

ra3xdh, 2025-05-07:
@ThomasZecha I have compiled the `dev/xml_devices` branch on my machine and can confirm that everything works. I will provide a more detailed feedback in the next weeks. I have created #1354, linked it to this issue and marked it as WIP.

ra3xdh, 2025-05-07:
It's need to implement an automatic way to dump the existing C++ devices to XML. Manual conversion would be almost impossible task.

ra3xdh, 2025-06-08:
@ThomasZecha , Could you check the reported issues on XML devices?

Radvall, 2025-06-08:
I have tested changes introduced in #1354. Below is my testing results:

## 1. Component highlight and simulation results

I noticed a strange selection behavior with these components — they highlight differently compared to standard components (which behave normally).
[image]
[image]

To test simulation functionality, I created a simple voltage divider circuit using resistors. The simulation ran without any errors, but no results were produced (e.g., no plots or values).

[image]

DC operating point simulation shows the same issue — no output.

[image]

## 2. Application Crash
After saving and closing the schematic created during voltage divider testing, reopening it causes QUCS-S to crash.

Similar crashes happen when when attempting to open certain existing schematics, for example `resistors.sch` from the examples:  
https://github.com/IHP-GmbH/IHP-Open-PDK/tree/main/ihp-sg13g2/libs.tech/qucs/examples However, other schematics from the examples are open and simulate without crash. 

Attachments:
My test project with the voltage divider - [test_XML_prj.zip](https://github.com/user-attachments/files/20646144/test_XML_prj.zip)

ra3xdh, 2025-06-13:
> The selection behavior will be examined.

I guess, the bounding box may be set wrong.

ThomasZecha, 2025-06-16:
Bounding box as well as property placement fixed by https://github.com/ra3xdh/qucs_s/pull/1410