ra3xdh/qucs_s #1050: Qucs Triangle Wave Example won't work Qucs-S

tomhajjar, 2024-11-08:
This may effect #1046

I have been having issues making I/V Triangle sub-circuits that work in ngspice and qucsator-rf.

1) netlist created for "Triangle_i_tran.sch" has an extraneous "-1" that causes ngspice errors. I flipped polarity of I1 and "-1" is still there. Send netlist to DuSpice after deleting -1 and simulation runs but data is garbage.

2) I went back to Qucs 0.0.20 and "Triangle_Wave_Generator.sch" works. Try Triangle_Wave_Generator.sch in Qucs-S, qucsator-rf and it has multiple errors.

[image]
[image]
[image]

[Waveforms_prj.zip](https://github.com/user-attachments/files/17682389/Waveforms_prj.zip)

ra3xdh, 2024-11-12:
The irect simulation error fixed by #1059.  The triangle wave simulation example works on my machine.
[image]

closed 2024-11-12 by ra3xdh

tomhajjar, 2024-11-12:
Have you updated the "nightly"? The latest from 35min ago hasn't fixed issues...

ra3xdh, 2024-11-12:
Everything should be updated automatically. The latest build from `current` branch works as expected.

ra3xdh, 2024-11-12:
Use the continuous release from 18:18UTC

tomhajjar, 2024-11-12:
Still having issue with "Triangle_v_tran_qucs.sch". I am checking my work in Qucs and Qucs-S