ra3xdh/qucs_s #1053: Trasnmission line S-Param not showing correctly for shorted load

iwbnwif, 2024-11-10:
In the following simulations, S[1,1] has a low value resistance as a load while S[2,2] has a short. The electrical length of the transmission line is approximately lambda/2 so there should be a complete circle for both simulations. This only occurs for loads approximately < 0.1R.

[image]

[smith_problem_0R.zip](https://github.com/user-attachments/files/17690717/smith_problem_0R.zip)

ra3xdh, 2024-11-10:
I confirm this bug. The solid line curve type is rendered correctly, but the circles is not.

tomhajjar, 2024-11-11:
Does it in ngspice as well.  You can vary Td over a wide range and plot looks similar. Only star spacing changes.

[image]
[image]

[S-parameter plot.zip](https://github.com/user-attachments/files/17694395/S-parameter.plot.zip)

closed 2025-08-12 by ra3xdh