ra3xdh/qucs_s #1068: TL071 in SpiceOpamp doesn't work

tomhajjar, 2024-11-17:
Not sure how it happened but the TL071 by Nazar Shtybel in SpiceOpamp doesn't work. The model generates many ngspice warnings and the model generates large errors when the feedback resistor values are >1k. For R=10k the simulation is totally wrong. 

I have many other models for the TL071 as well as othe devices and SpiceOpamp could use an update. 

[image]
[image]

[OpAmp_Library_Test_prj.zip](https://github.com/user-attachments/files/17791155/OpAmp_Library_Test_prj.zip)
[OpAmp_Spice_Models_prj.zip](https://github.com/user-attachments/files/17791157/OpAmp_Spice_Models_prj.zip)

ra3xdh, 2024-11-17:
The model is not damaged and works as expected. The warnings about `T_ABS` parameters of diodes may be ignored. I will clean the model in the library.

[image]

tomhajjar, 2024-11-17:
Try R1/R2=10k

ra3xdh, 2024-11-17:
Everything works as expected with 10k/10k. The model is not broken. I am using no compatibility mode. 

[image]

ra3xdh, 2024-11-17:
10kHz source also works as expected.

tomhajjar, 2024-11-17:
I'm not sure what is going on. With R1/R2=100k, output is shifted by +6 volts...

[image]

[TL071_Non_Inverter.zip](https://github.com/user-attachments/files/17791614/TL071_Non_Inverter.zip)

tomhajjar, 2024-11-17:
I used ngspice 38 and the issue is gone...

I will try to find what rev "broke" the model...

[image]

ivandi69, 2024-11-17:
I can reproduce it with the latest ngspice from git and it is most likely caused by ngspice commit [a43c6f4916ef68029e7c777bd1485cc87609751d](https://sourceforge.net/p/ngspice/ngspice/ci/a43c6f4916ef68029e7c777bd1485cc87609751d)

Note these warnings about resistor value too small:
.SUBCKT TL071_SN  1 2 3 4 5
...
R4 6 0 1E-12 TC=130e-2
R19 0 32 6E-9
R20 0 34 6E-9
[image]

tomhajjar, 2024-11-17:
Hard to understand the reasoning for those 3 resistors are in the netlist. All provide a path to "0" for sources. Author uses the same scheme in his other Spice models like NE5534..etc.

I used his Spice models because they modeled Vcc/Vee currents which was important for audio power amplifiers with OpAmps in a Booster/Bootstrap configuration.

tomhajjar, 2024-11-17:
Holger says...

"I have uploaded a fix to development branch pre-master-44."

ivandi69, 2024-11-17:
RESMIN has been changed to 1E-12. So there is still one warning, but it seems to work as expected:
[image]

tomhajjar, 2024-11-17:
There is no reason to have resistor values that small in the model especially since the Tempco parameters will be removed.

ra3xdh, 2024-11-18:
The problem was on Ngspice side. Closing this.

closed 2024-11-18 by ra3xdh