ra3xdh/qucs_s #1075: S-parameter File Handling Observation

tomhajjar, 2024-11-18:
@ra3xdh  @trans

For a Mini Circuits ADQ_22+ Hybrid coupler I was comparing how Qucs-S makes a sub-circuit from the S-parameter files versus how S2spice makes a netlist from the S-parameter file. Mini Circuits says the device needs a ground reference.

S2spice makes a netlist from the S-parameter file with 4-ports and a ground resulting in a sub-circuit with a ground.
Qucs-S makes a sub-circuit from the S-parameter file with 4-ports BUT without a ground.

Both sub-circuits work but if the ground is removed from the sub-circuit created from the S2spice created netlist, it doesn't work. 

[image]
[image]

[S-parameter_files_ngspice_prj.zip](https://github.com/user-attachments/files/17803674/S-parameter_files_ngspice_prj.zip)

tomhajjar, 2024-11-19:
Here is a S-parameter file project made for qucsator-rf. 

When Qucs-S is in "Qucsator mode", a 2-port S-parameter file whether placed into the schematic directly or made into a sub-circuit will ALWAYS have 2-ports plus the 3rd reference port.

When Qucs-S is in "ngspice mode" a 2-port S-parameter file placed directly into the schematic directly will ONLY have 2-ports.
A 2-port S-parameter file converted by S2spice will ALWAYS have 2-ports plus the 3rd reference port. It will be

[S-parameter_files_qucs_prj.zip](https://github.com/user-attachments/files/17822433/S-parameter_files_qucs_prj.zip)

transmitterdan, 2024-11-21:
I decided to make S2spice use the "reference" port because sometimes you may want that reference to be something other than "ground". Others have decided to force the reference of each port to be node 0 (aka "ground") so they don't require you to connect it on your schematic.  But the connection is there just hidden inside the subcircuit. I felt this was overly restrictive.

If one were to be pedantic, every port should have 2 terminals (v+ & v-) since S-parameters assume that is the case.  However, very few practical circuits actually have the references for all ports floating. That would make the schematic symbol larger and require numerous ground connections to be made to every port.

If you think there is something wrong with how S2spice works, please explain because when I read your messages, I don't see a specific action suggested.

tomhajjar, 2024-11-21:
No actions for now. 

I was making an observation that Qucs-S handles S-parameter files differently for ngspice than it does for qucsator or how S2spice handles it. Some might be confused that there is no gnd/ref. It will also preclude connecting an impedance to the ref port.

tomhajjar, 2024-11-21:
@ra3xdh @transmitterdan

Would adding User selectability be a major change? The device could default to no Ref unless the User makes a change.

closed 2024-12-29 by ra3xdh