ra3xdh/qucs_s #1076: Windows version of ngspice.exe is updated nightly but ngspice_con.exe is not

tomhajjar, 2024-11-18:
I was wondering why the ngspice "pre-44" nightly was not fixing my issue. Turns out the Windows version of ngspice.exe is updated nightly but ngspice_con.exe is not. Hopefully this will get rectified...

https://sourceforge.net/p/ngspice/discussion/127605/thread/4d19fbe8f8/?limit=25#a3c2

ra3xdh, 2024-11-19:
I have informed Ngspice team that `ngspice_con.exe` is necessary for the operation of the Qucs-S Windows version. Closing this.

closed 2024-11-19 by ra3xdh

ra3xdh, 2024-11-19:
The simulation may work, but the logs may not be shown. In the past `ngspice.exe` showed GUI window and didn't produce output in the STDOUT. The Qt communicates with the child process using the STDOUT/STERR. This makes impossible to obtain and analyze logs from GUI version. The `ngspice_con.exe` has the same behavior as the Linux version. You may check if the logs are written correctly when using `ngspice.exe`.

tomhajjar, 2024-11-19:
Confirmed ngspice.exe 43+ released October 18th, 2024 doesn't produce a log but the latest nightly ngspice.exe 43+ does.

I need to ask the ngspice guys...

tomhajjar, 2024-11-19:
Giles says the script to create the nightly was wrong. ngspice.exe is actually ngspice_con.exe.

He is looking to redo what files are created and maybe add more options and features.

 https://sourceforge.net/p/ngspice/discussion/127605/thread/d51ea6f6a3/?limit=25#9488

tomhajjar, 2024-11-20:
Windows 43+/pre 44 ngspice_con.exe has been updated.

Download and install "full nightly" from October. I create new folders for new revisions like C:\Ngspice-43+_64
https://ngspice.sourceforge.io/download.html

Download "partial nightly" from today and over-write directories/files in directory above.
https://github.com/gatk555/ngspice/actions/runs/11919116638

tomhajjar, 2024-11-25:
The ngspice nightly for Windows now includes ngspice.exe, ngspice_con.exe and all support files.

Windows is "Build others"

https://github.com/gatk555/ngspice/actions