ra3xdh/qucs_s #1077: Wrong steps and number of points in simulation dialogs after #1054

ivandi69, 2024-11-19:
This transient simulation setup generates wrong netlist with a negative tran step: .tran -0.001 0.001 0
[image]
[image]
AC linear sweep gives wrong number of points ex: 500.95:
[image]
Most likely all sweeps are broken. They worked fine since #615

ra3xdh, 2024-11-19:
@iwbnwif , Please have a look. Revert an old behavior implemented in #615 if necessary.

iwbnwif, 2024-11-19:
Thanks for reporting, I will look into this.

iwbnwif, 2024-11-19:
I think this is fixed by #1079, at least for the two reported issues. As far as I can tell, the behavior for both log sweeps and lin sweeps is essentially similar to the production 24.4.1 version; however, there may be some edge cases where the user inputs invalid values for log sweeps.

Also, I notice that the list 'sweeps' do not work properly in 24.4.1 or the new component dialog. Those problems are not addressed by this fix and I will open a separate issue after more investigation.