ra3xdh/qucs_s #1081: Single Frequency Analysis Status

tomhajjar, 2024-11-21:
It's been a long time since we discussed Single Frequency Analysis and I don't know the status.

I imported a newly made Qucs 0.0.20 project that was converted from QucsStudio. Qucs allow single frequency but Qucs-S greys out the menu settings. Is this normal?

[image]
[image]

[Combline_Filter_qucs_prj.zip](https://github.com/user-attachments/files/17839491/Combline_Filter_qucs_prj.zip)

ra3xdh, 2024-11-23:
> It's been a long time since we discussed Single Frequency Analysis and I don't know the status.

The single frequency sweep in SPICE is a hack. It requires vector assembling to work as expected. Otherwise you will get a set of 1x1 plots. This operation cannot be automatized. Write the script by hands if this simulation is needed. 

The single frequency and list sweep should work only when Qucsator is selected. 

> Qucs allow single frequency but Qucs-S grays out the menu settings.

This may be a consequence of #1054 The values are not grayed out in my case, but the *Apply* button doesn't work if *value* is selected. The *list* works as expected. Also the list preset value has disappeared. This may be misleading for the new users, because it's need to know the list syntax. Here is the screen recording.

[Screencast_20241123_090111.webm](https://github.com/user-attachments/assets/bf37f30d-58a5-444f-bdcd-ebb2f2a655c3)

The preset values from the old dialog:
[image]

@iwbnwif , Please have a look at this.

iwbnwif, 2024-11-23:
Yes, it is almost certainly a problem with the new dialog.

Tbh, I am not sure if the single 'value' option is needed. It is easy to just enter a single value in the list. At the moment, the either [100 MHz] or 100 MHz will work with the 'list' option to give the desired result.

iwbnwif, 2024-11-24:
> Qucs allow single frequency but Qucs-S grays out the menu settings. Is this normal?

This is because the file has the sweep type defined as `const` which is not currently implemented in Qucs-S. Instead, the sweep type should be `list` and have a single value, either alone or surrounded by square brackets.

iwbnwif, 2024-11-24:
> Also the list preset value has disappeared. This may be misleading for the new users, because it's need to know the list syntax.

I have restored the functionality whereby the start and stop values are used to create the list when the dialog is opened, or if they are changed at all. Conversely, providing the list is correctly formulated, the start and stop values will be updated from the list.

A slight change from 2.4.4 is that the dialog format for the list now requires the user to enter the square brackets. I think this is reasonable because it makes the list format consistent with how it is if you are editing on the schematic, although I accept it might confuse some users who were used to the old behavior.

ra3xdh, 2024-11-25:
> I would like to do some more refactoring on this class

Yes, renaming the properties is a bad design. But we must to keep the things as is. Introduction of new properties will break the schematic save/load. The schematic load relies on fields order, but not on fields name. The renaming to `Symbol` is a hack, because the property named `Symbol` doesn't go to the Qucsator netlist. The implementing of #974 will partially resolve this issue. I am intending to introduce some `Simulation/NoSimulation` flag. I am planning to start the implementation of #974 by the end of December or January.

ra3xdh, 2024-11-25:
Fixed by #1093

closed 2024-11-25 by ra3xdh