ra3xdh/qucs_s #1089: File not found Issue

tomhajjar, 2024-11-24:
@ra3xdh

Qucs-S displays ngspice errors and halts even when the netlists sent via DuSpice to ngspice work.

In the attached two projects the following occurs:

Ngspice started...
Error opening osdi lib "HICUML2V3.osdi": No such file or directory!
Error: Library HICUML2V3.osdi couldn't be loaded!

and

Ngspice started...
igbt3.osdi: no such command available in ngspice
Error opening osdi lib "igbt3.osdi": No such file or directory!
Error: Library igbt3.osdi couldn't be loaded!

The "no such file or directory" Error has happened many times over the past week and only goes away when I use DuSpice to send netlist to ngspice. Attached is two projects and four netlists.

HICUML2V3_ic_ib.cir is the raw unedited netlist
HICUML2V3_ic_ib_edit.cir is edited to get a plot

igbt3_ic_vg_DC.cir is edited to get a plot
igbt3_sub_ic_vg_DC.cir is edited to get a plot

[image]
[image]
[image]
[image]
[image]

[HICUM_IGBT.zip](https://github.com/user-attachments/files/17891745/HICUM_IGBT.zip)

ra3xdh, 2024-11-24:
It is not a bug. Qucs-S cannot resolve the relative path inside scripts (the *Nutmeg script* component). The netlist is placed in some cache directory, but the OSDI path is given relatively to schematic. It's need to implement a full Nutmeg language parser to implement paths resolution.  The specialized Verilog-A device component would also help. But I don't consider an extension of these features in the near future. You should always define the absolute path in scripts.

closed 2024-11-24 by ra3xdh

tomhajjar, 2024-11-24:
Files updated. 
Of interest to users is using sub-circuits to add symbols and hold the path to the osdi file

[image]
[image]

[Verilog-A_osdi.zip](https://github.com/user-attachments/files/17893512/Verilog-A_osdi.zip)

tomhajjar, 2024-11-25:
Here is a Verilog-A EKV3 project.
https://sourceforge.net/p/ngspice/discussion/120973/thread/415d5b0326/?limit=25

[Verilog-A_EKV3_osdi_prj.zip](https://github.com/user-attachments/files/17895762/Verilog-A_EKV3_osdi_prj.zip)

tomhajjar, 2024-11-25:
@ra3xdh 

In the attached EKV3 project, 2 of the example files can't work because ngspice cannot process my long PATH name in a Nutmeg Script because it has spaces in it. This is odd since ngspice handles the same long PATH with spaces in the Nutmeg Scripts used in the other 14 examples. Do I have a syntax issue?

"pmos_cov_vg.sch"and "nmos_cov_vg.sch" won't work but the 14 other do...

[image]
[image]
[image]