ra3xdh/qucs_s #1092: MSVC Build Testing Needs.

dsm, 2024-11-24:
Hi @ra3xdh, I manually build and create portable release using MSVC. There are two version one for Windows on ARM other for normal x64. Initially native build has less .dll compared to msys2 build. I need feedback If someone wants to test it. Because I want to add these builds options in our CI deployment.
 
[qucs-s_msvc_x64.tar.zip](https://github.com/user-attachments/files/17893670/qucs-s_msvc_x64.tar.zip)
[qucs-s_msvc_arm64.zip](https://github.com/user-attachments/files/17893671/qucs-s_msvc_arm64.zip)

```
sha256hashes:
qucs-s_msvc_x64.tar.zip: a2a3d21d148ef2c393b9f54720595f2b9474c82773bf4359d461f5b7670d9fb5
qucs-s_msvc_arm64.zip: 72bec6657e9199fbabf388a8eda161d75f42c09a1c072904f44426e3e219aa18

```

tomhajjar, 2024-11-25:
I did a quick test in Windows. It uses the same configuration file so it knows where qucsator-rf is and the project folder.

dsm, 2024-11-25:
Recently I solved a issue related to msvc build, Drag n Drop a components cause a crash. Qt behave differently sometimes, and MSVC handle C++ differently but I added some compiler flags to solve this but maybe there are unknown bug or misusage of qt. In my test there is no practical diff. Btw build size decreased and we have a windows on arm build, x64 version runs on windows on arm but prisim emulator not great.

dsm, 2024-11-29:
I tested Windows on ARM build on MacOS with VMware Fusion and works as expected.

dsm, 2024-12-02:
Hi, now you can able to download MSVC portable and setup in the actions menu, just click latest build and download(I hide in release.).

[image]

tomhajjar, 2024-12-02:
I tested a dozen projects, ngspice, xyce and qucsator. Everything seems OK.

Will qucsator-rf  eventually be included?

Can OpenVAF.exe be made to work without installing all the MS development tools? @kreijstal is working on a fork of OpenVAF.

#1083

Kreijstal, 2024-12-03:
> I tested a dozen projects, ngspice, xyce and qucsator. Everything seems OK.
> 
> Will qucsator-rf eventually be included?
> 
> Will this "fork" of Qucs-S allow OpenVAF.exe to work without installing all the MS development tools?
> 
> @Kreijstal is working on a fork of OpenVAF.
> 
> #1083

qucs-s doesnt need ms development tools.

dsm, 2024-12-03:
I just try other compilers to succesfully compile qucs-s, qucsator_rf can't build on msvc but I sent a PR to compiling qucsator_rf with native windows env using mingw don't need msys2 but can't build on msvc right now but I'll try my best.

tomhajjar, 2024-12-03:
@Kreijstal

OpenVAF needs MS tools to work under Qucs-S for Windows. I was asking if the MSVC Build of Qucs-S would also need MS tools loaded to use OpenVAF...

Below are the "tools" needed to compile va files using Qucs-S/OpenVAF on my Windows PC.

[image]

dsm, 2024-12-03:
Qucs-s build doesn't include any compiler this is just a build of qucs-s using msvc tool.

Kreijstal, 2024-12-03:
@tomhajjar old openvaf needed ms tools, openvaf-r can be built using msys2 and msvc (I haven't tested msvc builds but feel free to try)

ra3xdh, 2024-12-10:
The MSVC build didn't show a visible difference from MinGW build. I think this issue may be closed.

dsm, 2024-12-10:
Yes, I didn't find any issue related to MSVC tool. Thanks All. Closed.

closed 2024-12-10 by dsm