ra3xdh/qucs_s #1097: Moving/Placement of Objects in Schematic causes simulation to fail

tomhajjar, 2024-11-26:
An issue I have been having when working on Verilog projects is simulations failing after a simple cosmetic change to the schematic. 

Just moving almost any non device object on screen can cause simulation to fail. It causes me to click the simulation button after every change to see if simulation is still functional. If sim fails I click "Undo", save the schematic and give up.

I finally decided to document a specific schematic:

"npn_spar_NF.sch" works and "npn_spar_NF_test.sch" doesn't. Only difference is the S-parameter simulation box is 1 grid spacing lower. 

The osdi file must have a complete PATH.

[image]
[image]

[Verilog-A_HICUML2_osdi_prj.zip](https://github.com/user-attachments/files/17910722/Verilog-A_HICUML2_osdi_prj.zip)

tomhajjar, 2024-11-26:
@dwarning says Qucs-S automatically finds the osdi file, if it exists, and adds it with the PATH to the netlist.
https://github.com/dwarning/VA-Models/issues/9

I followed his recommendation and it worked...

dwarning, 2024-11-26:
I experienced this problem too some time ago.
Even it is solved by the automatic pre_osdi insertion, it will remain in case the user want use an dedicated osdi file in a different place without to have the file in the "working" directory.

ra3xdh, 2024-11-26:
I have tried to make some schematics from scratch using the recent version from Git and didn't observe this issue. Do you observe this with only one specific schematic or with every schematic? Are you using a MSVC build prepared in #1092 or usual MSYS2 build? Could you provide a screen recording to reproduce what exactly are you doing?

ra3xdh, 2024-11-26:
The error messages in the log indicate that there is something wrong with MSVC redist libraries. Probably the Ngspice some libraries are mixed with the libraries from the another installation because of `$PATH` issues. The netlists looks identical. The problem is not in the netlister.

ra3xdh, 2024-11-26:
I have tried to simulate the attached schematic. The issue still be "cannot reproduce". The both schematics work as expected after recompiling OSDI files and fixing the paths in the Nutmeg script. Then I tried to move different devices and schematic still work. I used Linux machine. I will try another Ngspice version later. I have found an error in your *Nutmeg script* component. The newline after the `pre_osdi` command is not allowed. The file name should follow the `pre_osdi` after space. I am wondering the simulation works. 

> @dwarning says Qucs-S automatically finds the osdi file

This feature works only for projects and doesn't works for schematics not belonging to the projects.

dsm, 2024-11-26:
> The error messages in the log indicate that there is something wrong with MSVC redist libraries. Probably the Ngspice some libraries are mixed with the libraries from the another installation because of `$PATH` issues. The netlists looks identical. The problem is not in the netlister.

MSVC build title is 24.4.1

ivandi69, 2024-11-27:
I stumbled upon similar issue when moving a component on the schematic caused simulation fail.
I narrowed it down to an easily reproducible example of two identical schematics. The only difference is the order in which the source and the resistor are put in the netlist. To "fix" the failing schematic one has to move the source. That will change the netlist
from failing:
```
V1 _net0 0 DC 0 SIN(0 1 1K 0 0 0) AC 1 ACPHASE 0
R1 0 _net0  1K tc1=0.0 tc2=0.0
```
to working:
```
R1 0 _net0  1K tc1=0.0 tc2=0.0
V1 _net0 0 DC 0 SIN(0 1 1K 0 0 0) AC 1 ACPHASE 0
```
Moving the resistor will cause the simulation to fail again.

Obviously this is a corner case and an issue with ngspice. The failing simulation log is:
```
Bprobe_int_r1Vref: no such command available in ngspice
Bprobe_int_r1power: no such command available in ngspice
.save: no such command available in ngspice
vcurr_r1:probe_int_n2:2__net0: no such command available in ngspice
vcurr_r1:probe_int_n1:1_0: no such command available in ngspice
Using SPARSE 1.3 as Direct Linear Solver
Warning: singular matrix:  check node probe_int_0_r1_1
```

I am not sure if Qucs-s can do anything about this issue, but we can remove the "cannot reproduce" label.

[test-pass.sch.txt](https://github.com/user-attachments/files/17934650/test-pass.sch.txt)
[test-fail.sch.txt](https://github.com/user-attachments/files/17934651/test-fail.sch.txt)

ra3xdh, 2024-11-27:
Yes, I can reproduce this now. The CLI Ngspice version also fails. The issues is not on Qucs-S side. The SPICE netlist syntax should not be sensitive to the devices order. This should be reported to Ngspice team. Here is the minimal netlist to reproduce this issue:

~~~
* Qucs 24.4.1  /tmp/test-fail.sch

.PROBE P(R1)

*R1 0 _net0  1K tc1=0.0 tc2=0.0 
V1 _net0 0 DC 0 SIN(0 1 1K 0 0 0) AC 1 ACPHASE 0
R1 0 _net0  1K tc1=0.0 tc2=0.0 

.control

ac dec 10 1 1k 
let power = R1:power
plot power

.endc
.END
~~~

This netlist is failing using the Ngspice-43. Swap the V1 and R1 line and the netlist will work. The bug appears only if the .PROBE directive presents in the netlist. 

Unfortunately there is no workaround on the Qucs-S side. There is no way to control the devices order in the netlist.

tomhajjar, 2024-11-28:
Is someone going to report this to the ngspice guys?

dwarning, 2024-12-01:
It is fixed by Holger's commit yesterday.
Thanks for reporting.

ra3xdh, 2024-12-01:
Closing as resolved.

closed 2024-12-01 by ra3xdh