ra3xdh/qucs_s #1101: Library with User selectable parameters

tomhajjar, 2024-11-28:
I'm trying to make a Library but I'm having issues adding User selectable parameters to a "Spice netlist" device.  I searched but can't find any example of doing this. Adding User selectable parameters to a "Spice library" device works BUT they can't be made into a Qucs Library.

First device is a PBRS 7 generator. Long term goal is to add  "Systems Analysis" to Qucs-S. This will also add value to Ideal.lib and Xanalogue.lib which I doubt many use.

[image]
[image]
[image]
[image]

[PBRS7_subcircuit.zip](https://github.com/user-attachments/files/17941789/PBRS7_subcircuit.zip)

ra3xdh, 2024-11-28:
I am aware about this limitation. SpiceFile device doesn't support parameters. The `.PARAM` will have no effect in this case. Wait until the SpiceLibComp extensions #944 will be implemented. There is no workaround.

closed 2024-11-28 by ra3xdh

tomhajjar, 2024-11-28:
Presently under the "Qucs way" of making sub-circuits, User definable parameters MUST be on both the schematic sub-circuit and symbol sections. AFAIK on schematic param=param and on symbol param=value.

Will this be the "new" method as well?

[image]
[image]
[image]

[Parameter_Passing_prj.zip](https://github.com/user-attachments/files/17950085/Parameter_Passing_prj.zip)