ra3xdh/qucs_s #1105: Quc-S is inserting ".PARAM Export = no" into the netlist

tomhajjar, 2024-11-30:
I don't see this happening in all the other cases of using ".PARAM" in a schematic but Quc-S is inserting ".PARAM Export = no" into the netlist causing ngspice to flag it as an error and halting.

I'm looking into using XSPICE for "Systems Analysis" and wanted to reproduce the ngspice XSPICE PLL example. Unfortunately this issue popped up.

For this specific case, "Phase_Freq_Det_test.sch" I'm trying to make a schematic of "test-f-p-det.cir" and trying to understand why it is so "complex". digital zero, digital one, d to a for plotting...etc. It seems XSPICE digital requires a lot of extra "support" items in a netlist.

[image]

[image]

[XSPICE_PLL_prj.zip](https://github.com/user-attachments/files/17966338/XSPICE_PLL_prj.zip)

ra3xdh, 2024-11-30:
@tomhajjar  Remove the "Put result in dataset" checkbox in the `.PARAM` properties dialog and everything will work as expected. This checkbox may be easily set by mistake.

[image]

@iwbnwif  Please have a look at this. The *Put result in dataset* should not be enabled only for Qucs equations. The .PARAM is never put in the dataset. And Nutmeg equations is always put in the dataset. SPICE handles the both automatically and there is no way to control this manually. The *Put result in dataset* is Qucsator only feature.

tomhajjar, 2024-11-30:
It was never checked. I tried checking and un-checking and the same result.

Like I mentioned this isn't an issue on all the other "digital" projects I have using .PARAM VCC=xx. It's unique to this schematic.

[image]

ra3xdh, 2024-11-30:
> It was never checked. I tried checking and un-checking and the same result.

I cannot reproduce this with my schematics. The error appears only if this checkbox is checked. Need further investigations.

tomhajjar, 2024-11-30:
What about the project I sent?

ra3xdh, 2024-11-30:
Yes, i can reproduce it. I have found that the `.PARAM` has extra `Export` property in the XML schematic file. This made the schematic broken. Please wait for fix.

iwbnwif, 2024-11-30:
> The Put result in dataset should not be enabled only for Qucs equations. 

Please can I check if this should say "should be enabled only for Qucs equations"?

I.e., no other forms of equations even need this checkbox

ra3xdh, 2024-11-30:
> Please can I check if this should say "should be enabled only for Qucs equations"?

Yes, this is a typo. This option should present only for Qucs equations.

tomhajjar, 2024-11-30:
How is it the schematic line format changed?

Old:
<SpicePar SpicePar1 1 60 350 -28 16 0 0 "vcc=5" 1>

New:
<SpicePar SpicePar1 1 20 250 -28 16 0 0 "vcc=5" 1 "Export=no" 0>

ra3xdh, 2024-11-30:
> How is it the schematic line format changed?

This is a bug. The *SpicePar* should not contain *Export* property.

tomhajjar, 2024-11-30:
Editing the schematic with a text editor is a short term solution that works.

iwbnwif, 2024-11-30:
@tomhajjar thank you for spotting this. I have put in a PR to hopefully fix this problem.

The line `export = no` might appear in the parameter list if you open a file that was created with this bug. However, you can freely delete the line everything should be okay from then on.

Please let me know if you find any other problems or this doesn't fix it.

iwbnwif, 2024-11-30:
@tomhajjar sorry slightly OT but when I run the simulation in your project the ngspice result set does not match with the graphs. I've noticed this with quite a few of the examples that you provide.

For example, I have no d_clk or d_out1 results, and I am wondering why that is?

[image]

tomhajjar, 2024-11-30:
I noticed this as well. There is always one missing vector. Many times I don't plot the vector and don't know it's missing.
In this one case it was "fixed" by adding a "dummy" line "dR = R+0". 

Now if I could only get my various phase-frequency detectors to work...

[image]

[PLL_CMOS_prj.zip](https://github.com/user-attachments/files/17967172/PLL_CMOS_prj.zip)

iwbnwif, 2024-11-30:
> Now if I could only get my various phase-frequency detectors to work...

Good luck! It's very useful that you are giving the new dialog a good workout. I hope it makes inputting and editing equations a little bit easier.

ra3xdh, 2024-12-01:
Fixed by #1106

closed 2024-12-01 by ra3xdh