ra3xdh/qucs_s #1108: Kicad 8 ngspise

mikhailmihalkov, 2024-12-01:
When installing the simulator in Linux, it does not work with the built-in simulator, Kicad, it is not possible to install it separately due to library dependencies in the system. The problem is old, maybe there is a solution for Linux?

ra3xdh, 2024-12-01:
There still be no solution. This is observed only on Ubuntu if KiCAD is installed via PPA. All software that requires true ngspice (not a library) become broken. Read the the thread #333 Read also: https://gitlab.com/kicad/packaging/kicad-ubuntu-builder/ngspice-kicad-pkg/-/issues/4 You may delete ngspice-kicad if the simulation in KiCAD is not required. You may also switch to KiCAD version from system Ubuntu repository. It is not affected by this bug. I have reported this issue to KiCAD team a year ago but there is no response so far. It was also reported before: https://gitlab.com/kicad/packaging/kicad-ubuntu-builder/ngspice-kicad-pkg/-/issues/2 You may try to raise the bugreport mentioned above to attract attention of the KiCAD PPA maintainers. 

I am closing this, because the problem is in the wrong configuration of the KiCAD PPA package. It cannot be resolved on Qucs-S side.  Also it was already reported as #333

ra3xdh, 2024-12-01:
Compile Ngspice by hands and installing it in the `/opt` will also help.

ra3xdh, 2024-12-01:
You may also try the Qucs-S version from Flatpak. It has embedded Ngspice and Qucsator. But Flatpak has many disadvantages because of sandboxing concept. They are: the fixed version of simulators and not possible to use XYCE and OpenVAF.

closed 2024-12-01 by ra3xdh