ra3xdh/qucs_s #1111: Digital Library only "partially" working after update

tomhajjar, 2024-12-03:
The Digital Library "74HCng_gates.lib" modified to work in ngspice by Holger is only "partially" working after an update. Only the 1st item in the Library, 74HC00 still gets "processed" properly by "Spice library". All others don't get assigned a Spice library "symbol". 

Oddly all sub-circuits still work even missing the Spice library symbol...

There are other Digital Libraries like the PSpice ones with similar structure/syntax of the .SUBCKT line like this.
[image]
[image]
[image]
[image]

[Digital_Analog_Mixed_prj.zip](https://github.com/user-attachments/files/17993639/Digital_Analog_Mixed_prj.zip)

tomhajjar, 2024-12-03:
Attached is a MOSFET with thermal project I did last month that now doesn't work. Unlike above where previously made sub-circuits still work, none work in this project.

[image]
[image]

[MOSFET_Si_thermal_prj.zip](https://github.com/user-attachments/files/17995664/MOSFET_Si_thermal_prj.zip)

tomhajjar, 2024-12-03:
Here is a GaN project using a device previously used in a KiCAD/ngspice example that doesn't work anymore.

I don't remember why I made a modified version of the GaN library...and why it works...

[image]
[image]

[GaN_thermal_prj.zip](https://github.com/user-attachments/files/17995952/GaN_thermal_prj.zip)

ra3xdh, 2024-12-03:
Fixed by #1113 This has appeared after #1074

PR #1113 (2024-12-03): Fixed parsing SPICE library
This PR fixes #1111
closed 2024-12-03 by ra3xdh