ra3xdh/qucs_s #1114: CI Windows issues

ra3xdh, 2024-12-03:
I have just noticed that the MSYS2 Windows target has been excluded from CI, but remains in deploy.yml. Also this target is marked as `continue-on-error` The MSYS2 Windows target (not MSVC) should be blocking. @dsm?

[image]

dsm, 2024-12-03:
It says 7 succesful, I think windows build also there, I didn't remove it but yes continue-on-error should be remove. if you looked latest pr for this check, I've also checked and windows build already there.

ra3xdh, 2024-12-03:
Yes, everything found. May be some web interface issue. Closing.

[image]

closed 2024-12-03 by ra3xdh