ra3xdh/qucs_s #1115: Digital Subcircuits with pull-up/dn resistors or vcc/gnd connections

tomhajjar, 2024-12-03:
@ra3xdh

I can't figure out how to make sub-circuits using the PSpice Digital libraries that have pull-up/dn resistors or vcc/gnd connections unless I add Pins for vcc/gnd to the sub-circuit so they can be connected in the main schematic. The vcc net or .PARAM vcc=5 entries didn't work. Making sub-circuits using PSpice Digital libraries with vcc/gnd pins looks odd since the devices themselves don't have a "physical" vcc/gnd...

My goal is to make a Library of higher-level digital devices.

[image]

[image]

[PLL_CMOS_prj.zip](https://github.com/user-attachments/files/17997563/PLL_CMOS_prj.zip)

tomhajjar, 2024-12-13:
@ra3xdh 

Adding a source to the subcircuit with U='vcc' seems to work. 

I haven't tried doing a pull-down. I assume a connection to ground would work.

[image]
[image]

ra3xdh, 2024-12-29:
Closing as answered.

closed 2024-12-29 by ra3xdh