ra3xdh/qucs_s #1117: Blue Diode in non-linear components menu can't change it's symbol

tomhajjar, 2024-12-04:
Blue Diode in non-linear components menu can't change it's symbol anymore.

Should the Red diode also have symbol options?

[image]

iwbnwif, 2024-12-04:
This is a result of 782f5929ff8519ee680ea8966b3ce008275d5b47  and is because some 'components' set certain property names to `symbol` in order to mask them. 

I need to work out which components mask properties in that way and only hide the `symbol` entry in those instances.

tomhajjar, 2024-12-04:
I never tried the "Populate parameters..." with the Blue diode so I didn't test it.

1) The Blue "Qucs" diode parameters don't match 1:1 with the ngspice diode parameters. What happens to the "un-used" parameters? I assume this diode only work with ngspice. If the schematic is opened in "qucsator" mode, does this "Blue" diode trigger an error?

2) The Red "Spice" diode can be assigned parameters that ngspice doesn't support but it's OK because ngspice flags and ignores them. 

3) Why wasn't the Red diode given the "Populate parameters..." feature? Can Red diode also have symbol options? Today you have to make a sub-circuit and give it the desired symbol.

[image]

[Rectifiers_prj.zip](https://github.com/user-attachments/files/18009418/Rectifiers_prj.zip)

ra3xdh, 2024-12-04:
>  I assume this diode only work with ngspice. 

Blue diode works with all simulation kernels. The incompatible parameters are ingnored while doing neltisting. The #974 will hide incompatible parameters from user. The Red diode cannot be even inserted in Qucsator mode. 

> Why wasn't the Red diode given the "Populate parameters..." feature?

This diode was designed to copy-paste the modelcard as is. This device is also needed for microelectronics PDKs. The *Fill from SPICE* option should be available only for blue devices.

tomhajjar, 2024-12-04:
@iwbnwif 

One requirement of "Blue" devices is the Reference Designation starts with a "fixed" Letter like "D" for the Diode and SHOULD have an underscore "_" at the end for proper device numbering.

A new User wouldn't know this. I wonder if a note should be in the device pop-up window?

[image]

[Rectifiers_prj.zip](https://github.com/user-attachments/files/18012048/Rectifiers_prj.zip)

ra3xdh, 2025-01-06:
I have no time to implement the #974 before the release, so I decided to add a temporary fix for this issue. See #1190

closed 2025-01-06 by ra3xdh