ra3xdh/qucs_s #1118: [ REQUEST ] adding Qt App style selection to settings.

dsm, 2024-12-04:
Hi @ra3xdh, I am currently working on to solve this feature request. Below are the tasks to be completed.

- [ ] add style selection combobox to application settings.
- [ ] add `-style` parameter to `launchTool ` to ensure tools have the same style with the qucs-s.
- [ ] refactoring some svg file to looks good on dark theme.

Thanks.

ra3xdh, 2024-12-04:
> Qt to converting all non-optimized SVG to Optimized SVG(Plain SVG) ?

The SVGs used for device icons and toolbar icons were optimized using the SVGO tool https://www.drupal.org/project/svgo The other SVGs need to be checked. 

>  refactoring some svg file to looks good on dark theme.

This may require to have two sets of images: for light and dark theme. The most of device icons are rendered automatically. It is not clear how to render the icon for the dark theme. The simple color inversion will not work. Probably you need to provide an options for wire color, device color, etc for light and dark theme.  This will require to refector the components, because the colors are hardcoded now.

dsm, 2024-12-04:
some SVGs has white background some has transparent option in tools as shown below tool screenshot. I'll remove it, and adding white-background in code as workaround but yes changing color of text and wire for dark theme.

[image]

ra3xdh, 2024-12-04:
Yes, the images in Active filter tool are not optimized.

dsm, 2024-12-04:
I don't remember but old issue has mentioned some svg icon has yellowish border artifact but actually svg has a border it should be remove manually.

```
<path d="M.5.5h127v127H.5z" fill-opacity="0" stroke="#fff" stroke-linecap="square" stroke-linejoin="round" style="paint-order:stroke fill markers"/>
```

[image]

ra3xdh, 2024-12-06:
Implemented by #1125.

closed 2024-12-06 by ra3xdh