ra3xdh/qucs_s #1119: Adding more symbols to Spice library menu

tomhajjar, 2024-12-04:
Can the Schottky diode symbol be added to the menu? SB5100 is a Schottky diode

[image]

ra3xdh, 2024-12-04:
Yes, I can upload additional symbols. Is this symbol in `schematic_symbols.sch`?

tomhajjar, 2024-12-04:
Yes. 

The latest symbol files are not included with distribution. All have been updated and tagged with date.

[image]

[Symbol_Creation_prj.zip](https://github.com/user-attachments/files/18011020/Symbol_Creation_prj.zip)

tomhajjar, 2024-12-04:
One requirement of "Blue" devices is the Reference Designation starts with a "fixed" Letter like "D" for the Diode and SHOULD have an underscore "_" at the end for device numbering. A new User wouldn't know this.

[image]

[Rectifiers_prj.zip](https://github.com/user-attachments/files/18011894/Rectifiers_prj.zip)

ra3xdh, 2024-12-29:
Fixed by #1161

closed 2024-12-29 by ra3xdh