ra3xdh/qucs_s #1120: Spice netlist works but Spice library doesn't

tomhajjar, 2024-12-05:
Spice library is "confused" trying to "process" VCO_MOS_sub.cir. It doesn't like the three subcircuits in the middle.

Subcircuit .subckt ro_vco
sub 1
sub 2
sub 3
.ends ro_vco

Spice netlist gets processed by ngspice which works.

[image]
[image]
[image]

[PLL_XSPICE_prj.zip](https://github.com/user-attachments/files/18017268/PLL_XSPICE_prj.zip)

ra3xdh, 2024-12-05:
The netlist contains the nested subcircuits. This syntax was not allowed by Ngspice in the past and Ngspice showed a warning and even may fail the simulation with the convergence error. Has something changed in the recent versions?

tomhajjar, 2024-12-05:
Unsure when ngspice allowed nested sub-circuits. The vco subcircuit "vco_sub.cir" has been in the ngspice XSPICE examples folder for a VERY long time. It is so old it was replaced by "vco_sub_new.cir' and hasn't been used for many releases of ngspice. It's actually an "orphan" sub circuit that was left in the folder. "vco_sub_new.cir' uses the ngspice "exclusive" "d_osc" function.

I would need to ask Holger when nested sub-circuits were allowed.

C:\Ngspice-43+_64\examples\xspice\pll

ra3xdh, 2025-01-19:
Fixed by #1211

closed 2025-01-19 by ra3xdh