ra3xdh/qucs_s #1121: dc bias doesnt rander values very well in Windows 11

jgardona, 2024-12-05:
Hello, good morning.

Dc bias calculation doesnt render correctly the values in the circuit. The values are rendered abouve the circuit wires, and it hasnt a background color(maybe same color of the circuit windows? ). In the Ampmeter, the same issue also happens. 
I think, the way that qucs studio works would be fit.

Im using windows 11 64x and qucs 24.4.1

ra3xdh, 2024-12-05:
Please submit a screenshot illustrating this issue. I didn't change the DC bias rendering in the recent release.

jgardona, 2024-12-05:
> Please submit a screenshot illustrating this issue. I didn't change the DC bias rendering in the recent release.

I im doing it now. Must install again.

jgardona, 2024-12-05:
Here it is:

in Qucs-S
[image]

[image]

in Qucs Studio

[image]

ra3xdh, 2024-12-05:
It is a long known issue. Qucs-S and old Qucs always rendered DC bias in a such way. May be adding a background to the DC bias result will improve the readability. We will consider making this modification. QucsStudio is maintained by another developer and the source code has diverged from Qucs and Qucs-S a years ago.

jgardona, 2024-12-05:
> It is a long known issue. Qucs-S and old Qucs always rendered DC bias in a such way. May be adding a background to the DC bias result will improve the readability. We will consider making this [...]

Yep. Using a background color in font(I think the same as the background of the board), will definitelly improve the readability.

dsm, 2024-12-11:
I made some changes if it's OK, I'll sent PR. I modified value 4 point precision(but can be set or adding settings.) and ex convert 1.33000mA to 1.33mA and 1.0000mA to 1mA

[image]

[image]

ra3xdh, 2024-12-11:
It may be useful to introduce a setting named like *Transparent DC bias lables ON/OFF*.

jgardona, 2024-12-11:
Its very good in my opinion. The colors are beautiful.

dsm, 2024-12-11:
I tested, here is the result. Yes, we can add a transparency settings.
Before:
[image]

After:
[image]

jgardona, 2024-12-11:
Thats way more suitable to read.

ra3xdh, 2024-12-16:
Fixed by #1130

PR #1130 (2024-12-16): fix #1121
Hi, this PR fixed #1121.
commit 8a9c6400 (2024-12-16): fix #1121
commit 9482caaf (2024-12-16): Merge pull request #1130 from dsm/current

fix #1121
closed 2024-12-16 by ra3xdh