ra3xdh/qucs_s #1127: Corrupted data file crashes Qucs-S

tomhajjar, 2024-12-07:
A PLL simulation went amok and consumed all my PCs resources so I was forced to stop the simulation. The resulting data file crashes Qucs-S when the schematic is opened. 

The schematic is "10M_PLL_dosc_vco.sch" The corrupted data file is "10M_PLL_dosc_vco.dat.ngspice" which I had to rename *.bad.

Why the simulation runs amok I cannot say. I've made many changes to try and speed simulation including replacing the vco and 10M ref with very simple models. 

[image]

[PLL_CMOS_prj.zip](https://github.com/user-attachments/files/18046380/PLL_CMOS_prj.zip)

ivandi69, 2024-12-07:
You have to add .OPTION INTERP to reduce the dataset size. And you have to increase the number of points to have more details.
[10M_PLL_dosc_vco.sch.txt](https://github.com/user-attachments/files/18049382/10M_PLL_dosc_vco.sch.txt)

[image]

tomhajjar, 2024-12-07:
The netlists sent to the DuSpice GUI don't have those changes. Do you know why the netlists work?

ivandi69, 2024-12-07:
KLU is supposed to be faster. Sparse works fine too. The netlists work in Qucs-s too. Ngspice runs fine and produces a huge plot. Qucs-s needs a lot of resources to process that huge plot.

tomhajjar, 2024-12-07:
Using DuSpice doesn't use up all 16GB of my RAM and crash. Qucs-S is using more RAM running the same simulation as DuSpice.  I have to use Task Manager to monitor resources and kill Qucs-S before I get a RAM warning.

For these PLL simulations:
After ngspice finishes, DuSpice plots almost immediately.
After ngspice finishes,  Qucs-S can take a huge amount of time writing from RAM to the HD before doing the plots.

tomhajjar, 2024-12-07:
Made progress. I accidently set the VCO output to 1 instead of 5 volts. Also made a narrow-band VCO that is 10MHz at 2.5 volts. PLL is hunting and locking over a narrow range. The project is good enough to use as an example.

Will do more comparisons with DuSpice.

[image]

[PLL_CMOS_prj.zip](https://github.com/user-attachments/files/18050221/PLL_CMOS_prj.zip)

ra3xdh, 2024-12-29:
The Qucs-S cannot look forward and predict the dataset size. It is on the user. This issue cannot be fixed on Qucs-S side.

closed 2024-12-29 by ra3xdh

tomhajjar, 2024-12-29:
The issue was the netlist sent to DuSpice didn't have the same effect on my PC.

ngspice takes the same time to do the simulation using Qucs-S or DuSpice. It's after the ngspice is done. DuSpice handles large data files more efficiently. It doesn't take minutes to pot the data like Qucs-S.

As I made modifications to the PLL circuit, the data files got smaller.