ra3xdh/qucs_s #1128: .IC menu doesn't seem to work

tomhajjar, 2024-12-09:
I can't get the .IC menu to work. The 'Name' box is "locked'. 

Also there isn't an 'Add' button to allow more entries.

[image]

dsm, 2024-12-09:
I think this is the solution for this bug. I think recent refactoring in componentdialog cause this.

```
diff --git a/qucs/components/componentdialog.cpp b/qucs/components/componentdialog.cpp
index 5290229..e4e575c 100644
--- a/qucs/components/componentdialog.cpp
+++ b/qucs/components/componentdialog.cpp
@@ -348,7 +348,7 @@ ComponentDialog::ComponentDialog(Component* schematicComponent, Schematic* schem
   componentNameWidget->setCheck(component->showName);
 
   // Try to work out what kind of component this is.
-  isEquation = QStringList({"Eqn", "NutmegEq", "SpicePar", "SpGlobPar"}).contains(component->Model);
+  isEquation = QStringList({"Eqn", "NutmegEq", "SpicePar", "SpiceIC", "SpGlobPar"}).contains(component->Model);
   hasSweep = QStringList({".AC", ".DISTO", ".NOISE", ".SW", ".SP", ".TR"}).contains(component->Model);
   hasFile = component->Props.count() > 0 && component->Props.at(0)->Name == "File";
 
@@ -358,8 +358,8 @@ ComponentDialog::ComponentDialog(Component* schematicComponent, Schematic* schem
   // for a given simulation type. Then only create the valid widgets fo
   // sweepParams[".AC"] = QStringList({"Type", "Start", "Stop", "Points"});
 
-  paramsHiddenBySim["Export"] = QStringList{"NutmegEq", "SpicePar", "SpGlobPar"};
-  paramsHiddenBySim["Sim"] = QStringList{".AC", ".DISTO", ".SP", ".NOISE", ".TR", "Eqn", "SpicePar", "SpGlobPar"};
+  paramsHiddenBySim["Export"] = QStringList{"NutmegEq", "SpicePar","SpiceIC", "SpGlobPar"};
+  paramsHiddenBySim["Sim"] = QStringList{".AC", ".DISTO", ".SP", ".NOISE", ".TR", "Eqn", "SpicePar","SpiceIC", "SpGlobPar"};
   paramsHiddenBySim["Param"] = QStringList{".AC", ".DISTO", ".SP", ".NOISE", ".TR"};
 
   // Setup the dialog according to the component kind.

```

[image]

ra3xdh, 2024-12-10:
Yes, this is consequence of #1054 @iwbnwif Please have a look.

iwbnwif, 2024-12-10:
> I think below patch is the solution for this bug. I think recent refactoring in componentdialog cause this.

@dsm  Yes, the patch looks good to me, thank you.

iwbnwif, 2024-12-10:
Are there any other components in the Spice netlist sections group that should also be treated in the same way as equations etc.?

dsm, 2024-12-10:
yes you can look the spice netlist sections, you need to look at the differences between the old and new version of qucs-s if old qucs-s dialog has add button but don't equation, I think you can implement a way of adding a new table element.

tomhajjar, 2024-12-10:
Some menus like for example, Spice library or formulas would benefit from move-up, move-dn buttons. 

When I don't like the order or make a mistake I am forced to exit Qucs-S, open the schematic/subcircuit in a text editor and fix the order/mistakes.

[image]

ra3xdh, 2024-12-11:
Fixed by #1129

closed 2024-12-11 by ra3xdh

tomhajjar, 2024-12-11:
Spice Library subcircuit Properties could use the feature. I end up using a text editor to "clean-up" the order of the entries.

Just today I added new entries to a subcircuit and they end up on the bottom which put them out of order.

[image]

ra3xdh, 2024-12-11:
> Spice Library subcircuit 

There is no such device. Do you mean *Subcircuit* or *Spice Library Device*?

> Spice Library subcircuit Properties could use the feature. 

Probably you meant the dialog in subcircuit symbol editing mode. It is another dialog not related to component properties. It is `id_dialog.cpp`. This dialog would benefit from redesign too.

tomhajjar, 2024-12-11:
Spice Library Device. 

The .ID dialog is tortuous to use. It's faster to use a text editor to edit or add entries. Having entries out of order can be confusing. 

Some of the XSPICE digital devices have a lot of entries. Until you do a simulation, you don't know which entries need to be in the .ID. Another option is to be able to change the item view defaults.