ra3xdh/qucs_s #1132: Clock frequency using Qucs Rectangular Source versus Spice Pulse Source

tomhajjar, 2024-12-12:
I made two clock sources models. One using Qucs Rectangular Source and the other using Spice Pulse Source. I can't figure out why the one made with Qucs Rectangular Source generates the wrong frequency.

Since both use the Spice Pulse function why is the one I made using "Qucs" wrong? Is the mapping of Qucs parameters to Spice correct?

[image]
[image]

[Waveforms_prj.zip](https://github.com/user-attachments/files/18105223/Waveforms_prj.zip)

ra3xdh, 2024-12-12:
Everything is correct. The period for Qucs source is calculated as `PER=TH+TL+TR+TF`. If gives `2*0.5/1e7+1e-9+1e-9=1.02e-7` The calculation in the CIR file is different. Please fix your equations in the subcircuit to take into account TR and TF.

closed 2024-12-12 by ra3xdh

tomhajjar, 2024-12-12:
Actually they were both wrong. I had to fix clock.cir as well...

[image]

[Waveforms_prj.zip](https://github.com/user-attachments/files/18114485/Waveforms_prj.zip)