ra3xdh/qucs_s #1138: XSPICE Digital nodes won't Plot

tomhajjar, 2024-12-15:
Qucs-S has an issue plotting voltages from XSPICE digital nodes. In ngspice's version of XSPICE, component digital outputs switch "modes" if terminated with a resistor and insert an "Auto-Bridge" with analog output with default vcc=3.3 Volt (unless a .param vcc=xx is used). 

ngspice data format changes if a node is terminated versus un-terminated. Qucs-S plots terminated nodes properly but doesn't plot un-terminated "digital" nodes. Qucs-S doesn't process the data file properly for unterminated nodes.

I made some test schematics and netlists. The plotting issue doesn't happen with DuSpice.

https://sourceforge.net/p/ngspice/discussion/120973/thread/64cc577dd4/?limit=25#3eb9

[image]
[image]
[image]
[image]
[image]

[XSPICE_plotting_issue.zip](https://github.com/user-attachments/files/18140945/XSPICE_plotting_issue.zip)

tomhajjar, 2024-12-19:
@ra3xdh 

I made a project consolidating the XSPICE "digital/mixed" plotting issues I found so far.

Qucs-S is compressing the time-base of plots of XSPICE "digital" nodes by an order of magnitude, like 100:1. DuSpice confirms the issue is with Qucs-S not ngspice.

To get around time-base compression I use the ngspice "trick" of terminating digital nodes which creates an "auto-bridge". UNFORTUNATELY ngspice can only handle this feature under cases that are VERY hard to quantify. Too many times adding a resistor termination so one can plot the "digital" node causes the ngspice analysis to "blow-up" creating plots with HUGE voltage spikes (many KV!!!) instead of the expected "square" waveforms. DuSpice confirms the issue is with ngspice not Qucs-S.

[XSPICE_plot_issue_prj.zip](https://github.com/user-attachments/files/18200737/XSPICE_plot_issue_prj.zip)

ra3xdh, 2024-12-19:
Thanks for reporting. I will look at this issue in January. I have never tried to plot the XSPICE digital output using Qucs-S and cannot comment anything on this.

tomhajjar, 2024-12-19:
Can you explain the Subcircuit Port properties Menu in relation to XSPICE? I have never used this menu for anything other than Name and Number of a Port for XSPICE or "analog" subcircuits. 

Does changing Type or XSPICE Type do anything other than changing the Port Symbol?

[image]

ra3xdh, 2024-12-19:
> Can you explain the Subcircuit Port properties Menu in relation to XSPICE? 

This feature has no effect on simulation. It was required for the XSPICE Code Model synthesizer. See https://mos-ak.org/lausanne_2017/presentations/T03_Kuznetsov_MOS-AK_DATE_2017.pdf The build of CodeModels outside of the Ngspice source tree is not possible anymore. And therefore I decided to remove this synthesizer. See #204. The extra properties on ports also need to be removed, but it is required to investigate that the schematic save/load will be not broken.

gatk555, 2024-12-20:
Am I correct to think that QUCS-S works with ngspice, by invoking it as a program ("console mode") and saving the results in a SPICE raw file?  And that the raw file is explicitly created with "write", and the alternative "batch mode" option for creating it is not used?

If that is correct, the problem seems almost certain to be that QUCS-S is plotting all vectors against the default scale.  XSPICE digital nodes have their own time scales that should be used in plotting: if ff1_q is such a node, then its scale will be called ff1_q_steps.  Such nodes can also be recognised by the "dims=" flag in the header, which gives the actual length of the data vectors.

So the obvious fix would be to parse such files and plot correctly.  The only disadvantage is that such code must be duplicated in each program that acts as a simulator front-end.  An alternative would be to create a filter utility to "flatten" a raw file, so that all vectors share a common scale,  Or it might be done within ngspice - it could even be a command script.   A disadvantage is that a flattened raw file may be much larger that the original, but contains no new information.

ra3xdh, 2024-12-24:
> And that the raw file is explicitly created with "write", and the alternative "batch mode" option for creating it is not used?

Yes, Qucs-S writes the SPICE dataset using the Nutmeg `write` command in the `.control` section.

ra3xdh, 2025-01-06:
Setting the PlotVs helps to see the correct result:

[image]

[image]

tomhajjar, 2025-01-06:
What am I doing wrong?

[image]

ra3xdh, 2025-01-06:
I didn't find an easy way to eliminate the line to (0,0). The Ngspice inserts trailing zeros in dataset to balance the vector lengths.

tomhajjar, 2025-01-06:
Not a big deal. Getting plots to work was the big one. I can stop using so many resistors and D-A Bridges.

The resistors can cause problems for ngspice. The "Auto-Bridge" built into digital nodes can "misbehave" in some cases

ra3xdh, 2025-01-06:
I am working on solution. The next release should contain a fix for this bug.

ra3xdh, 2025-01-07:
The fix provided by #1194. The problem with trailing zeros is also fixed by this PR.

closed 2025-01-07 by ra3xdh

tomhajjar, 2025-01-07:
Looks good so far.

PR #1194 (2025-01-07): Fix plotting XSPICE digital nodes
This PR fixes #1138. Plotting digital nodes works correctly now. 

[image]