ra3xdh/qucs_s #1140: Problem with Qucs-S with Ngspice

Radvall, 2024-12-16:
I tried to simulate the following circuit with the OpAMP LM358 from Libraries -> SpiceOpamp.

[image]

For simulation I use Qucs-S-24.4.1 with ngspice installed from https://github.com/ra3xdh/qucs_s/releases/download/24.4.1/Qucs-S-24.4.1-setup.exe

The default path for ngspice is shown in the figure below:

[image]

I found an old discussion about ngspice's location on Windows here: https://sourceforge.net/p/ngspice/discussion/127605/thread/4c758f87/ and downloaded ngspice from sourceforge  and unpacked it in C:/Spice64/.

The circuit works as expected.

[image]

The executable location is shown in the figure below.

[image]

dsm, 2024-12-16:
can you uninstall and install the latest continuous build and test is again please.

dsm, 2024-12-16:
I found a problem I can provide a fix.

dsm, 2024-12-16:
ngspice needs this file `"Spice64\share\ngspice\scripts\spinit"` but we didn't include `\share\ngspice\scripts\` folder to qucs-s I'll add this folder to release thanks.

tomhajjar, 2024-12-16:
@Radvall 

The installation instructions for Windows are obsolete. I put ngspice in folders labeled by revision. I have multiple versions of ngspice installed and change the Simulator Settings for ngspice when I want to use a different version.

The use of .spiceinit in Windows should be done by placing the statement on the schematic. 

Alternatively a .spiceinit ASCII file can be placed in the ngspice bin folder or Windows User folder. Unfortunately that specific .spiceinit will be used every simulation which can cause some simulation to fails. I only put a .spiceinit in the ngspice bin folder when running ngspice outside of Qucs-S. I then delete it when done. 

[image]
[image]
[image]

dsm, 2024-12-16:
putting inside .spiceinit and sim should work.

```
if $?xspice_enabled
 codemodel ../lib/ngspice/spice2poly.cm
end
```

tomhajjar, 2024-12-16:
Since .spicinit isn't needed for this Library, why was it brought up?

[image]

dsm, 2024-12-16:
Error on line:
  a$poly$e.xop1.egnd %vd [ _net1 0 _net2 0 ] %vd ( xop1.99 0 ) a$poly$e.xop1.egnd
 MIF-ERROR - unable to find definition of model a$poly$e.xop1.egnd
    Simulation interrupted due to error!
    
    it gives this error

dsm, 2024-12-16:
@tomhajjar in model file uses POLY(), if I am not wrong it depends Xspice

tomhajjar, 2024-12-16:
I have multiple XSPICE projects and none need a .spiceinit. What am I missing?

dsm, 2024-12-16:
can you remove spinit file in ngspice's share/scripts folder ?

tomhajjar, 2024-12-16:
The only .spiceinit files in the Windows ngspice release are in the examples folder.

In Windows, .spiceinit is only used if placed in 3 directories, "working", User and ngspice\bin

ra3xdh, 2024-12-16:
Fixed by #1141

closed 2024-12-16 by ra3xdh

dsm, 2024-12-16:
@tomhajjar 

it uses this file if I remove this then sim failed.

[image]

tomhajjar, 2024-12-16:
So "spinit" not "spininit"

dsm, 2024-12-16:
if ngspice don't find this file but you put .spiceinit in schematics. it gives this warning.

`Warning: can't find the initialization file spinit.`