ra3xdh/qucs_s #1142: Qucs-S won't allow single pin sub-circuits unless a variable is also on the .subckt line

tomhajjar, 2024-12-16:
For Qucs to accept a device netlist with a single pin there must be "additional text" on the .subckt line

.subckt example1 1 2                ;OK
.subckt example2 1                   ;BAD
.subckt example3 1 Value=1     ;OK

I'm forced to add "dummy" pins to single pin netlist that don't connect to anything. If single pin netlist has a variable, Qucs accepts it and the sub-circuit works. 

This is especially true with XSPICE components with no ground or power pins.

ra3xdh, 2024-12-29:
Could you submit a single pin netlist example for testing? Which device (*Spice File* or *Spice library Device*) are using with single pin device?

tomhajjar, 2024-12-29:
Here are examples using netlists with an XSPICE code model to make subcircuits of "digital" pull-up/dn. I had to add a "dummy" pin because Qucs-S wouldn't accept the netlists with 1 pin. I probably could have added parameter passing but that seemed confusing.

[XSPICE_Digital_prj.zip](https://github.com/user-attachments/files/18269867/XSPICE_Digital_prj.zip)

Here is examples where single pin subcircuits work to make "analog" pull-up/dn. Qucs-S didn't care the subcircuits had 1 pin.

[Digital_LV_prj.zip](https://github.com/user-attachments/files/18269886/Digital_LV_prj.zip)

ra3xdh, 2024-12-29:
Fixed by #1162

closed 2024-12-29 by ra3xdh