ra3xdh/qucs_s #1143: Qucs-S keeps crashing...

tomhajjar, 2024-12-16:
The latest nightly keeps crashing and I cannot pin down a specific trigger. 

Crashes occur during saving a file, running simulation...etc.  

Crashes also occur when opening a schematic containing subcircuits that were edited. In some case the subcircuit gained or lost pins.

I have also reported previous releases crashing when Qucs tries to open a corrupt or null data file.

As far as I know, it hasn't corrupted schematics...

tomhajjar, 2024-12-16:
I created a project using a schematic "d_Clock_2_tran.sch" that causes Qucs to close/crash. Schematic was created to chase down a bug in ngspice.

Open schematic, enable clk2 & nutmeg2, disable clk1 & nutmeg1 - run simulation.. qucs might crash

If crash:

Open schematic, disable clk2 & nutmeg2, enable clk1 & nutmeg1 - run simulation.

Qucs crashes/closes every time BUT it writes the correct data to the file and the data is visible when schematic re-opened. The schematic enable/disable selections made before crash weren't saved so they revert to schematic on HD.

[image]

[XSPICE_crash_prj.zip](https://github.com/user-attachments/files/18158067/XSPICE_crash_prj.zip)

ra3xdh, 2024-12-17:
Fixed by #1144

commit c04edc69 (2024-12-17): Merge pull request #1144 from ivandi69/1143-fix

Fix #1143
commit c74ae467 (2024-12-17): Fix #1143
Fixes a double free regression caused by #1139