ra3xdh/qucs_s #1148: How to make a Library from subcircuits with parameter passing like Xanalogue.lib

tomhajjar, 2024-12-18:
Does anyone know how @MikeBrinson made Xanalogue.lib or have the original project?

I can't figure out how to make a Library using "SPICE netlist" subcircuits with parameter passing like done in Xanalogue.lib. I have a number of new libraries to make but all require parameter passing.  

I made CSWO_lib.sch and CSWO_net.sch using CSWO.cir but. Only CSWO_lib.sch passes parameters but it cannot be made into a Library.  

I made XA_CSWO_tran.sch using Xanalogue.lib and it allows passing parameters.

When I compare XA_CSWO_tran.cir to CSWO_net.sch, there is a seemingly redundant subcircuit in CSWO_net.sch that "overrides" out_low=-2 out_high=2

[image]
[image]
[image]

[XSPICE_Analog_prj.zip](https://github.com/user-attachments/files/18188250/XSPICE_Analog_prj.zip)

ra3xdh, 2024-12-19:
The Xanalogue.lib was probably made by hands. Unfortunately I don't posses the sources and cannot give an exact answer.

ra3xdh, 2024-12-29:
The requested feature will be implemented as the part of #846. Closing a separate issue.

ra3xdh, 2025-01-08:
The requested feature implemented by #1197

closed 2025-01-08 by ra3xdh