ra3xdh/qucs_s #1150: FFT in mixed signal Analysis

tomhajjar, 2024-12-21:
Is there a way to do an FFT on a specific vector in without using a Nutmeg Script?

The Qucs-S FFT and Four analysis won't let you pick a specific vector so analysis fails when "digital" data is encountered.

[image]
[image]

gatk555, 2024-12-23:
I think part of the problem may be similar to #1138: the digital nodes cause ngspice to return a raw file that QUCS-S cannot parse.  And the sluggishness could be caused by ngspice eating all your system memory.  There may be a single fix for both: put "esave none" into a .control section of the the netlist.  Or "pre_esave none" you do not fully control the order of commands.

tomhajjar, 2024-12-23:
What Qucs-S "Spice netlist section" function should I use for "esave none" or "pre_esave none"?

I use Task Manager to monitor resources and the biggest offender is Qucs-S. ngspice finishes "most" XSPICE mixed simulations in a reasonable time and then Qucs-S spends 10x more "processing" the data. If I use DuSpice there is no delay once ngspice has finished, plotting is immediate.

ra3xdh, 2024-12-29:
Try to use the Nutmeg script simulation for this simulation type.

closed 2024-12-29 by ra3xdh