ra3xdh/qucs_s #1153: How to setup default language(eg. Chinese) when compiling?

shixiaogegeya, 2024-12-23:
I have compile(cmake and make) successfully in default branch.
But, when i run "./qucs-s", the default language is English.
I can't find how to setup the language to Chinese in Cmakelists.

shixiaogegeya, 2024-12-23:
When I set File-->Application setting, my terminal shows
"
xiaogege@xiaogege-VMware-Virtual-Platform:~/桌面/qucs_s-24.4.0/build$ ./qucs/qucs-s 
Warning: QSocketNotifier: Can only be used with threads started with QThread (:0, )
MESA: error: ZINK: failed to choose pdev
libEGL warning: egl: failed to create dri2 screen
Warning: Empty filename passed to function (:0, )
Warning: Empty filename passed to function (:0, )
Warning: Wayland does not support QWindow::requestActivate() (:0, )
Warning: Empty filename passed to function (:0, )
Warning: Empty filename passed to function (:0, )
Warning: Empty filename passed to function (:0, )
Warning: Empty filename passed to function (:0, )

"

ra3xdh, 2024-12-23:
I guess you have missed the installation step. The translation is loaded from `*.qm` files. The application must be installed properly to find the translations. The libraries and tools also requires an installation for the proper operation. Try to install the application using `make install` step. You have to set the installation directory to where you have write permission using the `-DCMAKE_BUILD_PREFIX` switch for cmake step. Then run the Qucs-S from the installation directory, not from the build directory.

ra3xdh, 2024-12-26:
No feedback from issue reporter. Assume this was resolved by the proper installation. Closing.

closed 2024-12-26 by ra3xdh