ra3xdh/qucs_s #1158: Can be added inductor and capacitor non-ideal models ?

dsm, 2024-12-27:
Hi, LTspice has these options to adding easily non-ideal model parameters for inductors and capacitors to simulate frequency response more accurately. 

[image]
[image]

ra3xdh, 2024-12-27:
It is already implemented a long time ago. There exists INDQ and CAPQ devices. The resistance is defined using the Q-factor which is more common for RF circuits analysis. See #143

dsm, 2024-12-27:
Oh ok but is it possible to add non-file spice device to schematics?

dsm, 2024-12-27:
@tomhajjar thanks for the example

andresmmera, 2024-12-27:
You can download s-par and pspice data from Murata and Coilcraft sites.

I've collected some models here [1]. Take a look to [2] too, I have some examples there.

[1] https://rfdesigntools.pythonanywhere.com/libraries

[2] https://rfdesigntools.pythonanywhere.com/MurataRF_filter

dsm, 2024-12-27:
@andresmmera thanks for the examples. I'll look into these.

tomhajjar, 2024-12-27:
@andresmmera

Can you include some of the subcircuits used to make the libraries?

andresmmera, 2024-12-27:
You can download the libraries from the links above.

I simply copied the pspice of the parts I needed from the vendor site and pasted that into a library.

tomhajjar, 2024-12-27:
You didn't make a subcircuit for each device using "Spice netlist" and then use the Create Library function?

tomhajjar, 2024-12-27:
I have a number of Libraries to make that use parameter passing (74xx, XSPICE analog/digital, PLL..etc) and it seems manually making the Libraries is the only option for now. Is there an example somewhere on how to do this? 

The thought of manually making hundreds of devices into Libraries is not pleasant....

ra3xdh, 2024-12-29:
Closing as answered.

closed 2024-12-29 by ra3xdh