ra3xdh/qucs_s #1159: Rectangle cannot be selected after rotation

tomhajjar, 2024-12-29:
Either select rectangle and rotate OR copy rectangle and rotate copy. 

Once rectangle is rotated it cannot be selected OR edited.

[image]

[rectangle_rotate.zip](https://github.com/user-attachments/files/18268244/rectangle_rotate.zip)

adrienluitot, 2025-01-15:
I can confirm I have the same bug in 24.4.1

ra3xdh, 2025-01-16:
Fixed by #1206

tomhajjar, 2025-01-16:
The issue is only "1/2" fixed...

One the rectangle is rotated, it can be selected and moved BUT it cannot be "edited". The 4 corner points are "fixed".

ra3xdh, 2025-01-17:
> Once the rectangle is rotated, it can be selected and moved BUT it cannot be "edited".

Fixed by #1208

closed 2025-01-17 by ra3xdh