ra3xdh/qucs_s #1160: PSpice Digital Logic Library and Qucs-S components

tomhajjar, 2024-12-29:
I am slowly adding more "PSpice" digital components as well as clock sources, pull-up/down's, resistor termination packs..etc. 

All the components use the .Param vcc=xx. No provision has been made to pass parameters to the PSpice devices.

Looking forward to any input from your students.

I haven't tried making subcircuits to implement the "LTspice" method of plotting digital timing diagrams.

[image]
[image]
[image]

[Digital_LV_prj.zip](https://github.com/user-attachments/files/18268380/Digital_LV_prj.zip)

tomhajjar, 2024-12-31:
Having bad luck getting the PSpice CD4000 Library to work. Made 5 subcircuits and only 2 worked...

[Digital_CD_prj.zip](https://github.com/user-attachments/files/18278721/Digital_CD_prj.zip)

ra3xdh, 2025-01-04:
Thanks! I will upload the three libraries (CD4000, 74HC, 74LV) as is. The ICs presented in libraries are the most used in analog circuits design.

tomhajjar, 2025-01-04:
I only made partial libraries. There are more device subcircuits I made recently that can be added to them.

The subcircuits can used as-is for now.

tomhajjar, 2025-01-04:
What do you want to call each Library?

Initially I used Digital_HC.lib, Digital_LV.lib...etc.

Is that acceptable?

ra3xdh, 2025-01-04:
Yes, the library names are acceptable.

ra3xdh, 2025-01-04:
Libraries uploaded by #1182

tomhajjar, 2025-01-04:
Lastly. 

Since all these "Digital" Libraries are based on XSPICE, there needs to be either a "Shared" Library containing the Sources, I/O, pull-up/dn..etc OR each Library needs them added. 

I could make a new XSPICE "Components" Library containing them.

ra3xdh, 2025-01-04:
The better solution will be to add a separate library (For example Digital_AUX or another name ) with pullup/down logic high/low etc. This library will be common for all three libraries.

tomhajjar, 2025-01-04:
All the XSPICE devices I made require parameter passing so making Libraries
is impossible until Spice Library device gets updated or some automated way
to make Libraries is created. There are dozens of new devices subcircuits.

ra3xdh, 2025-01-04:
> All the XSPICE devices I made require parameter passing

This feature is planned as the part of #944 But it will be implemented later this year. Definitely not in this release cycle.

ra3xdh, 2025-01-04:
The #1182 merged and this issue is closed. The #1034 remains open.

closed 2025-01-04 by ra3xdh

tomhajjar, 2025-01-05:
Latest version of the Digital Libraries on 1/04/25.

Digital_CD4000 -> Digital_CD

[Digital Libraries.zip](https://github.com/user-attachments/files/18308945/Digital.Libraries.zip)