ra3xdh/qucs_s #1164: Path issues when moving files between projects

tomhajjar, 2024-12-29:
Made new subcircuits and test files in one project folder and test schematics and subcircuits don't work when moved to different project folder. 

Qucs looks in "bin" folder for subcircuit.

[image]

tomhajjar, 2024-12-29:
Copied files once again and problem seems to be gone. Also downloaded latest nightly.

Still a strange error message if it couldn't find a file/subcircuit/netlist...

ra3xdh, 2024-12-30:
I didn't change anything related to file paths processing in the latest build.

tomhajjar, 2024-12-30:
To run the "simple" schematic requires all the subcircuit schematics and their netlist files. I can only assume one netlist file was missing.

I have never seen that error message before so it was confusing...

PS

I had issues with other simulations as well. I was forced to "refresh" each subcircuit to get simulation to run. Refresh means open subcircuit, choose same netlist again, save and close. No changes are made...

closed 2025-01-01 by ra3xdh