ra3xdh/qucs_s #1171: .MEAS Usage in Qucs-S

tomhajjar, 2025-01-01:
.MEAS is used extensively in LTspice and there are some examples in ngspice but I haven't found any in Qucs-S.

I want to measure the phase between three voltages and then display it on the schematic. Can someone point out the changes to make to the script in this simple 3-phase digital clock analysis?

Future work will include multiple examples of DC\AC\TRAN\SP analysis using .MEAS.

[image]

[Digital_Measure_prj.zip](https://github.com/user-attachments/files/18286007/Digital_Measure_prj.zip)

ra3xdh, 2025-01-01:
The `.MEAS` directive is not directly supported by Qucs-S. Maybe it could be used with scripting. I never try this directive. It is not clear how to add its support. I am not considering implementing this feature in the near future.

ra3xdh, 2025-01-01:
Moved to Discussion/Ideas. Probably someone will add the `.MEAS` support.