ra3xdh/qucs_s #1175: Convert Qucs-s *.sch files into some readable format for EESchema.

0range0range, 2025-01-02:
Hi all,

Is there a possibility to open Qucs-s schematic in EEschema? Are there working converters?

I only saw this one, but it was not updated for ages:  https://github.com/Valber/qucs2kicad

The goal is to do a layout after simulation without redrawing schematics in EEschema again.

Additional possibilities with indirect conversions or some other ways are also appreciated.

At the moment import dialog in EESchema includes the following: 

* Altium (*.SchDOC)
* CADSTAR (*.csa)
* Eagle XML (*.sch)
* EasyEDA std files (*.json)
* EasyEDA pro files (*.epro, *.zip)
* LTSPICE schematic files (*.asc)

What would creating such a converter take if one starts from scratch?

ra3xdh, 2025-01-02:
I didn't see any converters from Qucs SCH to KiCAD. The project from @Valber seems to be abandoned and not operational anymore. The conversion is technically possible, but there is no developers to implement this. Remember that Qucs-S is not supported by any organization (like KiCAD) and there is no full time developers in the project. You have to ways to implement this option: add converter yourself or find a developer to implement it. The starting point may be implement something like `Component::dumpKiCAD()` method.

ra3xdh, 2025-01-02:
On the other hand remember that the 99% of the PCB designs from KiCAD cannot be directly simulated. It is often required to split schematic into smaller parts and add supplementary devices to have success with analog simulation. The Qucs<-->KiCAD converted will have a very limited application only for some restricted class of electronic devices.

ra3xdh, 2025-01-02:
The more actual task is to add link to microelectronics CAD software like KLayout. Unlike the system level PCBs, the most of IC designs could be seamless transferred from layout to simulator and back.

ra3xdh, 2025-01-02:
Summarizing all above I am transferring this issue to Discussion/Ideas. May be reopened if one will find a developer for this task.