ra3xdh/qucs_s #1177: Debugging and Ability to see the Real netlist

tomhajjar, 2025-01-03:
Ran into a problem caused by a typo but the error messages from ngspice reference nodes and subcircuits hidden from the User. This is especially true for schematics containing subcircuits and Library devices.

To see the REAL netlist requires running ngspice.exe in which brings up it's console and typing commands to dump variables and the REAL netlist so the Error messages make a lot more sense. 

At a minimum the ability to see the REAL netlist would be a big help.

1) Use the commands 'display' and 'edisplay' to see what vectors are available in the current plot.
2) Use command 'listing e' (e for expanded) to see what netlist is finally simulated in ngspice after all the nested subcircuits have been inserted and expanded.
https://sourceforge.net/p/ngspice/discussion/120973/thread/73d9549257/?limit=25#a46f

[image]

ra3xdh, 2025-01-03:
The requested feature is not possible to implement within Qucs-S. Closing issue.

tomhajjar, 2025-01-03:
@ra3xdh 

Holger says just insert the following line into the netlist:

listing e > filename

I made the attached netlist. I haven't tried doing it within Qucs-S yet. I'm unsure how to place a "command" into the control loop.

[netlist creation.zip](https://github.com/user-attachments/files/18300206/netlist.creation.zip)

tomhajjar, 2025-01-03:
Found a solution using Nutmeg Script. Files are saved to spice4qucs folder.

[image]

[netlist-variables file save.zip](https://github.com/user-attachments/files/18301193/netlist-variables.file.save.zip)

ra3xdh, 2025-01-04:
Moved to Discussion/Ideas. Probably someone will implement it. I have more important task for this and next release cycles.