ra3xdh/qucs_s #1183: 555 XSPICE library device and subcircuits stopped working but netlists work with DuSpice

tomhajjar, 2025-01-04:
Something changed in ngspice or Qucs-S because the 555 XSPICE library device and the original subcircuits stopped working but netlists made from the same schematics work with DuSpice. All original netlists from 2023 also work.

555_XSPICE_lib_Oscillator.sch BAD
555_XSPICE_lib_Oscillator.cir  GOOD

555behav_Oscillator.sch BAD
555behav_Oscillator.cir  GOOD

555behav_PWM_10_90.sch BAD
555behav_PWM_10_90.cir  GOOD

555behav_PWM_90_10.sch BAD
555behav_PWM_90_10.cir  GOOD

[image]
[image]

[555 Timer_examples_prj.zip](https://github.com/user-attachments/files/18308386/555.Timer_examples_prj.zip)

ra3xdh, 2025-01-05:
Remove the DC simulation from the schematic. The Ngspice tries to find DC OP solution and fails. I am getting the same error with CLI version. The DC OP fails with 555 IC XSPICE model, but TRAN works. This case cannot be processed automatically. The Qucs-S cannot predict at which circuit DC OP will fail. 
[image]

closed 2025-01-05 by ra3xdh

tomhajjar, 2025-01-05:
A note in the Library might help a new User.